tsduck/tsduck #962: [Q]

sergeyshulga, 2022-03-18:
Hello. Is it possible to make TSDuck Srt plugin accept multiple connections in listener mode? Now its allow only one connection.
Thank you.

lelegard, 2022-03-18:
Do you refer to the srt input plugin or output plugin?

Do you mean multiple simultaneous connections or multiple sequential connections?

sergeyshulga, 2022-03-20:
SRT output plugin in listener mode.  Multiple simultaneous connections.
Multiple sequential connections working with --multiple key, as i understand.

lelegard, 2022-03-22:
> SRT output plugin in listener mode. Multiple simultaneous connections.

This is not possible. Only RIST has built-in support for multiple simultaneous outputs. Using SRT, this must be manually achieved inside the application, here `tsp`. The `tsp` processing model is one input / one output through simple plugins.

> Multiple sequential connections working with --multiple key, as i understand.

Exactly.

dutchboontje, 2022-04-25:
Hint:  You could have a look to sls (SRT-Live-Server)