tsduck/tsduck #964: [Q] About IAT increase in streams on some CPU

Koipse, 2022-03-18:
Greetings all!

Thank you for this wonderful toolkit! 

I noticed some interesting issue with different CPUs. I can be wrong with my inkling, so I need your help to understand what is going on with streams and find possible ways to fix it.

For example:

We have some stream in TSDuck with next settings where we merged some stream to NULL stream:

 tsp --bitrate 5000000 --buffer-size-mb 10.0 --max-flushed-packets 7 --max-input-packets 7 --receive-timeout 10000 -I null -P regulate -b 5000000 -P continuity --fix \
 -P merge "tsp --buffer-size-mb 10.0 --receive-timeout 10000 -I ip -l ${in_iface_ip} ${in_mcast} -P continuity --fix -P sdt -c -n 'Some_Channel' -s 1 -p 'Some_Provider'" \
 -b 5000000 --no-smoothing --acceleration-threshold 1500 --transparent --joint-termination --max-queue 3000 \
 -P pcradjust -b 5000000 -P regulate --pcr-synchronous \
 -O ip -l ${out_iface_ip} --ttl 128 ${output_channel}

We use server with Intel(R) Xeon(R) CPU E5-2667 v2. The stream works almost well except one moment (maybe you will have some comment how to do this stream better: delete, add or change some parameters?). After the streams worked some period, IAT has anomaly increase. Here is our streams monitoring system. You can see that sometimes the IAT increased, but we can't understand a source of this problem. The problem is gone when we restart the stream:

Our system:
[image]
Bridge tech:
[image]

I guess that it can be some issue with CPU compatibility or something like this. We use TSDuck 3.29-.3.30 on server with CPU Intel(R) Xeon(R) CPU E5-2680 v2 2.80GHz and don't notice issues like it in the pictures, but we notice issue with PCR. Are there some commands how to correctly recompute it?

If you want, we can provide for you this streams for analyzing.

Thank you!

lelegard, 2022-03-18:
What does IAT mean in that context?

Koipse, 2022-03-18:
> What does IAT mean in that context?

It means distance between frames (Or it can name jitter). Normal value need to be less that 40 ms. In visual monitoring, the video is strobing and freezing. And we don't know what to do...

Koipse, 2022-03-30:
Interesting thing: when we deleted "-P pcradjust -b 5000000 -P regulate --pcr-synchronous" the problem disapear. 

Close the ticket.

closed 2022-03-30 by Koipse