tsduck/tsduck #966: [FEAT] Make EXT-X-BITRATE key on HLS output media playlist optional

George-, 2022-03-21:
#### Is your feature request related to a problem?

The HLS (output) plugin generates a media playlist file with the EXT-X-BITRATE key for bitrate indications for media segments. 
This causes the media playlist to be rejected by YouTube lives HLS ingest feature, a similar test I ran with ffmpeg produced a media playlist file without the EXT-X-BITRATE key (this was the only difference) which was accepted. 

#### Proposed solution:
Add a flag to the HLS output plugin to make this key optional on the media playlist. Per the RFC specification this key is meant to be optional in a media playlist.

lelegard, 2022-03-22:
Hi @George- 

The option `--no-bitrate` has been added.

George-, 2022-03-24:
Thanks @lelegard! That is great 👍 I've noticed a separate issue which is preventing ingest to YouTube regarding HLS segment names, I've created #968 for that