tsduck/tsduck #967: [FEAT] pcrextract plugin to only log two values: PID and Value

moving-digital, 2022-03-21:
Hi Thierry @lelegard 

#### Is your feature request related to a problem?
Yes, related to a problem. I need to extract in real-time the PTS value from a TS for few days, therefore I am using the `pcrextract` plugin. 
This gives me the following data: PID, Packet index in TS, Packet index in PID, Type, Count in PID, Value, Value offset in PID, Offset from PCR.
The challenge: I can't restart the plugin, therefore the "Count in PID", "Packet index in TS" and "Packet index in PID" values increase super rapidly, which I want to avoid.

#### Proposed solution:
Would it be possible to add an option to `pcrextract` plugin such as `--value-only` in order to provide only two values: PID and Value.
This should work in conjunction with `--pts` and `--good-pts-only`.

#### Alternatives:
No alternative.

#### Additional information:
Using the latest TSDuch 3.31-2713.

lelegard, 2022-03-22:
> The challenge: I can't restart the plugin, therefore the "Count in PID", "Packet index in TS" and "Packet index in PID" values increase super rapidly, which I want to avoid, as they will never loop (unlike the PTS) so will reach the limit of the integer at some point.

All these values are represented as unsigned 64-bit integers. The one which increases the most rapidly is "packet index in TS". At 100 Mb/s (never heard of a faster TS), there are 66489 packets per seconds. We will reach the highest unsigned 64-bit integer value after 17,340,033,759,070 seconds or 200,694,835 days or 549,848 years. Should we worry about that?

Note that this evaluation has been done at the very beginning of the design of TSDuck, around 2005, and I decided to implement all packet counting operations on 64-bit integers and never worry about integer overflow.

moving-digital, 2022-03-22:
Thank you so much, @lelegard !

So sorry for having opened a ticket for that...

closed 2022-03-22 by moving-digital