tsduck/tsduck #968: [FEAT] Unique HLS segment filenames in --live mode

George-, 2022-03-24:
#### Is your feature request related to a problem?
HLS Segment filenames in --live mode are a rolling list of filenames, for example, segment0, segment1, segment2 etc, which are re-written to each time a segment is updated. This is not compatible with YouTube's HLS ingest for live streams, as it requires a unique segment filename for every segment: https://developers.google.com/youtube/v3/live/guides/hls-ingestion#media_segments 

#### Proposed solution:
Please can a flag be added to the HLS (output) plugin to make filenames unique, e.g. --unique-segments, rather than re-using the same segment naming structure (today in my setup it recurses through segment0, segment1, segment2, segment3, segment4, etc).

It would be sufficient to just continue incrementing the segment filenames and let the media playlist track which segments are being used in the live stream. If there are no knock on effects perhaps there is no disadvantage to making this default behaviour when using the plugin in --live mode. 

#### Alternatives:
A segmenter plugin to just generate TS segments, and then I could create the HLS implementation I need via a script, this could offer more flexibility to other people but requires work to manage files which the HLS plugin already does in TSDuck today.

lelegard, 2022-03-28:
> YouTube requires each segment in an HLS stream contains both PAT & PMT, and these should be the first two TS packets in each segment,

This is not a YouTube requirement, this is [specified in the HLS standard](https://datatracker.ietf.org/doc/html/draft-pantos-hls-rfc8216bis-10#section-3.1.1) and this is consequently [what is done in TSDuck](https://github.com/tsduck/tsduck/blob/75b64750aa87a543982b9c90048029918b56fb68/src/libtsduck/plugins/plugins/tshlsOutputPlugin.cpp#L287).

Apart from any consideration from standards, this is simply a matter of logic. Any packet before the PAT and PMT would be useless in an SPTS file.

Additional notes:
- The standard incorrectly assumes the PMT fits into one TS packet. In practice, this is the case with all simple streams as commonly found in basic Internet streaming. However, when grabbing a complex broadcast stream, the PMT may be larger. Any decent TS player should demux sections from the streams according to MPEG rules and never assume that a section always fits into one single TS packet.
- With the `hls` output plugin from TSDuck, the output segments are split from the input stream, after going through all plugins. If the input does not start with a PAT and PMT, the first output segment will start with the beginning of the stream, not a PAT/PMT pair. For subsequent segments, the `hls` output plugin always keeps a copy of the last PAT and PMT and adds a copy of them at the beginning of each segment.

lelegard, 2022-03-31:
In case the start of the first segment is an issue, I added the option `--align-first-segment`. Using this option, all TS packets before the first PAT, PMT, video PES packet boundary (and optionally I-frame) are dropped. Thus, the start of the first segment is HLS-conformant, at the expense of losing some initial TS packet, your choice.

closed 2022-09-23 by lelegard