tsduck/tsduck #969: Analyze Plugin output to udp socket [FEAT]

Desmoss900, 2022-03-29:
In implementing a MPEG-TS streaming solution for my private use I would like to be able to send the information generated by the Analyze plugin to an external UDP listener (or when starting a tsp chain from a os.Exec() call in a golang program).  I know this is possible in a bash script by writing to a pseudo device (/dev/udp/{address}/{port}) but this is a shell feature and is not available in golang.  With this enhancement, the analyze output could be stored in a database or collated with other information (i.e. stats from ffmpeg transcoding instance in an I/O chain) to give a complete overview of the stream health.

would love to see 

--output-file udp://{address}:{port} 

or

--output-udp {address}:{port}

Thoughts?

Mark

lelegard, 2022-03-29:
Since this type of usage is typically for automated processing, I assume that the JSON format is more appropriate than the large human-readable table.

In that case, an option `--json-udp address:port` seams more appropriate. There is already a `--json` option for JSON text file and a `--json-line` option for one JSON line in the log.

Desmoss900, 2022-03-29:
Hi Thierry,

Many thanks for your quick answer...agree with you and I have been using the --json output format when writing to a file...issue is when the plugin is writing to the file it locks and cannot be read by another process until the file lock is released (I had been polling the file every two seconds but ran into the locking issue numerous times)...with the output being sent to a udp address:port, it is then "fire and forget"...if the external udp listener does not receive the packet for some reason or other the last know info can be used in lieu...

--json-udp address:port is a great idea!!

Quick question...is the --json-line needed in addition or is it in lieu of --json?

Mark

lelegard, 2022-03-31:
Option `--json-udp` has been added. It comes with related options `--json-udp-local` and `--json-udp-ttl`.

Thanks to code reuse, these options were automatically added to `tsanalyze`, `tscmp`, `tsdektec`, `tsxml`, plugins `analyze`, `splicemonitor`.

Options `--json`,  `--json-line` and `--json-udp` are independent. Especially since the latest update; it is possible to set them independently and the JSON text is sent to all specified destinations, output file and/or log line and/or UDP datagram.

Desmoss900, 2022-03-31:
Fastastic!!!! I will give it a try right away :)

Desmoss900, 2022-03-31:
Works like a charm!!!! You are a star!!!

closed 2022-03-31 by Desmoss900