tsduck/tsduck #972: [Q] PMT video/audio description

Koipse, 2022-04-07:
Greetings!

First, thank you for your awesome toolkit!

I have read the documentation but didn't understand the next thing: is it possible to add PMT video/audio description by **-P pmt?**

I know that this function has --add-pid-registration, --set-data-broadcast-id, etc, but i can't understand the next:

When use --add-pid-registration and trying to add the hex description to PID, in result we have not that expect.

For example, we have some stream from ffmpeg on input without any VIDEO/AUDIO description. Added:

--add-pid-registration 49/0x060101 

In output, we have next information:

VIDEO: descriptor:  **050400**060101

And what is "05040"? Is that ISO/IEC 14496  scene/object description? Do we have any ways to delete this part? Or this function not for this? If yes, guess that need to create some "pmt".xml file with descriptions? 

Thank you for attention!

lelegard, 2022-04-08:
Hi @Koipse 

There are many ways to add "video/audio description", depending on what the system, operator or receiver expects. 

The option `--add-pid-registration 49/0x060101` adds a "registration descriptor" on PID 49 into the PMT. I have no idea whether this specifies some video format in some specific environment. 0x050400060101 is the content of the registration descriptor, including tag and length.

There are many ways to alter the PMT using the `pmt` plugin but you need to precisely know what you need to do at low level inside the PMT. If there is no suitable option for some very specific operation or unusual description, `--patch-xml` is the killer option.

TSDuck is a low-level tool. It is flexible but the price to pay is to precisely understand the MPEG and related standards (DVB and the like). Especially the structure of the PSI/SI tables must be fully understood.

Koipse, 2022-04-26:
Thank you @lelegard!

I deal with this task when saw some examples in documentation!

closed 2022-04-26 by Koipse