tsduck/tsduck #973: [Q] How to define AAC_Descriptor correctly

Desmoss900, 2022-04-11:
Hi,

I am trying to define the AAC_Descriptor in an xml file for injection via tsp and I cannot get the analysis plugin to report the stream properties set by the profile_and_level attribute...here is my pmt.xml file:

`<?xml version="1.0" encoding="UTF-8"?>
<tsduck>
  <PMT version="4" current="true" service_id="0x14" PCR_PID="0x96">
    <component stream_type="0x1b" elementary_PID="0x96">
      <stream_identifier_descriptor component_tag="0x9" />
    </component>
    <component stream_type="0x6" elementary_PID="0x97">
      <stream_identifier_descriptor component_tag="0x7" />
      <ISO_639_language_descriptor>
        <language code="eng" audio_type="0x0" />
      </ISO_639_language_descriptor>
      <DVB_AC3_descriptor component_type="0x0" bsid="0x8" mainid="0x0" />
    </component>
    <component stream_type="0xf" elementary_PID="0x98">
      <ISO_639_language_descriptor>
        <language code="eng" audio_type="0x0" />
      </ISO_639_language_descriptor>
      <AAC_descriptor profile_and_level="0x5b" SAOC_DE="false" AAC_type="0x8" />
    </component>
  </PMT>
</tsduck>`

no matter what value I enter in the AAC_Desciptor profile_and_level field I always see 

`HE-AAC Audio (eng, Audio layer 0, dual channel)`

in the audio description of the analysis output.  I know that the HE AAC stream is 3/2 channels, 48kHz 640kb/s and if I set the profile_and_level value to 91 (0x5b) as stated in ISO/IEC 13818-1:2015/Amd.4:2016  Table 2-72 – MPEG-4_audio_profile_and_level assignment values "0x5B High efficiency AAC profile, level 5" it has no affect on the analysis output...I would expect the analysis output to reflect the value that is set in the xml file.

Any tips or suggestions?

Mark

lelegard, 2022-04-14:
Hi @Desmoss900 

Thanks for reporting this. This seems to be an old issue in TSDuck (the TS analysis command was the first one to be developed, primarily for me to learn MPEG TS formats...) It was never reported so far.

The description "Audio layer 0, dual channel" is not taken from a descriptor. It is directly extracted from the PES data stream, independently of the signalisation. MPEG-2 Audio and AAC share the same stream id in the PES header (unlike other types of audio). In the early days of TSDuck, before AAC, this stream id was used for MPEG-2 audio only. Consequently, when such a PES stream was found, it was immediately interpreted as MPEG-2 audio and the audio format (here "Audio layer 0, dual channel") was extracted from the PES data. Of course, here the PES data are AAC, not MPEG-2 audio, and this interpretation is incorrect.

This needs to be changed but this is not trivial since the PES demux which reports the audio format from the content of the PES stream and the section demux which extract tables such as the PMT are independent. I need to think about it.

Additional remark: Stream type 0x0F seems incorrect for an AAC stream, it should be 0x1C. AAC is ISO 14496-3. According to ISO 13818-1, section 2.4.4.9, table 2-34, some values for the stream type in the PMT:
- 0x0F : ISO/IEC 13818-7 Audio with ADTS transport syntax
- 0x11 : ISO/IEC 14496-3 Audio with the LATM transport syntax as defined in ISO/IEC 14496-3
- 0x1C : ISO/IEC 14496-3 Audio, without using any additional transport syntax, such as DST, ALS and SLS

Desmoss900, 2022-04-15:
HI Thiery,

Just a quick shout out and THANK YOU to you for developing TSDuck!!!! I have been trying to get an IPTV streaming solution for my personal use at home up and running for many years now where I can stream my media files like a normal TV station...I have tried all the "usual" solutions (vlc, ffmpeg, etc.) and there have always been issues one way or another...TSDuck has been my life saver on this endeavor (even though I took me 2 weeks to find all the correct parameters to get a stable stream running longer than 24 hours)!!!! With my setup I can have a stream running 14 days non-stop without any issues (I haven't tested longer than 14 days yet as I am busy starting and stopping streams constantly during development)!!! I am in the process of migrating by bash script collection to a real micro-services based application on a Kubernetes cluster (so another thank you for doing the work on getting TSDuck working on Alpine Linux for the use in containers) using Go as the programming language (I have the requirement to learn Go for my "day job" so I thought this would be the perfect opportunity to learn the lang. a well as implement a somewhat robust solution for my IPTV ideas...I am not using the C++ bindings yet as that is quite a bit advanced for me at this point in time...just relying on os.Exec for tsp) 🥇 🥇 🥇 

Many thanks for the information...it is not critical for my application...just trying to get the "nuts and bolts" down correctly.  I use the analysis plugin on my transport streams to report the stream properties in real time (hence the request for the udp output of the analysis information 👍 ). I know that all my streams have an AAC audio track (as I transcode the files so all of them have the same properties) and was hoping it would report the channel layout correctly as it does for the AC-3 tracks...I will just "hardcode it" for now.

Also, I tried your recommendation to change the stream_type to 0x1c in the pmt.xml file and none of my playout devices (vlc, ffplay, kodi, iPhone (HLS), browser DASH, DuneHD STB) recognized the stream type as being an audio track with AAC audio...when I do a tsanalysis of the ts fie generated by my encoder (ffmpeg), I see the stream type being reported as 0x0f so that is why I used that value in the PMT file.

I will be posting some questions around the best way to implement an EIT injection strategy in the coming days.

Many thank again and regards,

Mark

lelegard, 2022-04-15:
> I tried your recommendation to change the stream_type to 0x1c in the pmt.xml file and none of my playout devices (vlc, ffplay, kodi, iPhone (HLS), browser DASH, DuneHD STB) recognized the stream type as being an audio track with AAC audio

That was a mistake I removed. 0x1C is for HE-ACC, 0x0F is for AAC. In broadcast TV streams, AAC has never been used in practice (at least I didn't see any), they moved from MPEG-2 audio to AC3 or HE-AAC. So, my first thought was HE-AAC.

Desmoss900, 2022-04-15:
Hi Thiery,

Agree with the broadcast community not using the „regular“ AAC but the streaming community definitely does (especially when RTMP is in the mix)…I generate my TS streams with both AC3 and AAC…this way another microservice can ingest the multicast stream and output a HLS or DASH stream easily 😁👍🏻

Regards,

Mark

Don‘t take life too seriously...no one gets out alive anyway....

lelegard, 2022-04-18:
Hi @Desmoss900 

Could you provide a small extract of your TS file? A few seconds should be sufficient. If too big to be attached to a comment here, please upload it to some file transfer site such as wetransfer.com or equivalent.

I plan to fix this and I would like to have some reference data to test.

Desmoss900, 2022-04-18:
Hi Thierry,

Am traveling at the moment and return Wed.  I will send you a file snippet as soon as I am able…if that is too long a wait, the easiest way to recreate it is with ffmpeg…simply encode with h.264 video, AC3 track and AAC track (I use libfdk_aac and not the built in aac codec)…mpeg_ts to pipe to tsp input for eventual file writing.

Regards,

Mark

Don‘t take life too seriously...no one gets out alive anyway....

lelegard, 2022-04-18:
> the easiest way to recreate it is with ffmpeg…

That's right. Done, do not worry about providing the file.

lelegard, 2022-04-18:
Hi @Desmoss900 

I fixed the identification of AAC, E-AAC and HE-AAC streams in the analysis.

In your specific context, I still see a problem in your PMT:
~~~
<component stream_type="0xf" elementary_PID="0x98">
  <AAC_descriptor profile_and_level="0x5b" SAOC_DE="false" AAC_type="0x8"/>
~~~

The stream type 0xf identifies "ISO/IEC 13818-7 Audio" (Table 2-34 of ISO 13818-1), meaning MPEG-2 ACC, not HE-AAC. On the other hand, in the AAC_descriptor, AAC_type 0x8 is "reserved for future use" (Table 26 of ETSI EN 300 468). More generally, it seems to me that the optional AAC_type in the AAC_descriptor should be omitted for AAC and meaningful only with HE-AAC. At least all documented values refer to HE-AAC.

Did you set AAC_type 0x8 on purpose?

Desmoss900, 2022-04-18:
Hi Thierry,

I was playing around with multiple values and could get none to work hence my original post…I will need to check what my original descriptor values were. I agree with your supposition on aac_type after reading the etsi spec 👍🏻…the fact that it was not reporting correctly was not due to the descriptor being incorrect 🤣🤣which it also was 🤣🤣🤣

Regards,

Mark

Don‘t take life too seriously...no one gets out alive anyway....

Desmoss900, 2022-04-21:
Hi Thierry,

updated my tsduck containers today and the analysis is showing MPEG-2 AAC Audio 👍🏻🤣 I am wondering if is possible to get the details of the stream similar to the AC-3 details (num of channels, layout, sample rate, etc) or is that information not easily obtained from the stream?

I did remove the AAC_type from the descriptor as recommended 👍🏻

Regards,

Mark

Don‘t take life too seriously...no one gets out alive anyway....

Desmoss900, 2022-04-21:
> This kind of information is extracted from the binary audio or video stream. We need to implement the right parser for each codec and detect the right information. This is not trivial. Over time, I took the time to implement that on MPEG-2 Audio, AC-3, MPEG-2 Video, AVC, HEVC and VVC. Doing the same thing on AAC and HE-ACC is not something I am ready to invest in at this stage.

Hi Thierry,

I fully understand 😁 As a workaround I track which file is currently active and use mediainfo stored in a db to pull the audio and video details 👍🏻

Regards,

Mark

Don‘t take life too seriously...no one gets out alive anyway....

closed 2022-09-21 by lelegard