tsduck/tsduck #975: [Q] Seems PCR/PID packets mixed on ip/multicast streams

dutchboontje, 2022-04-15:
Hi,

I am still testing your (marvellous piece ;) of software.

It came to my attention that in udp/multicast streaming it seems to occour a lot of 'Continuity check fails'

As a source I am using SRT, but HLS also has the same problem.
( even worse, becaus of the spiking, which are lots bigger )

So, I am trying to pin-point where the problem is.
At this stage I see some PKT shuffling... anyway have a look below.

I can not decide of this corrolated to the content or the processing of the content.

Last weeks I have been testing in linux/os performance.
At that level I do not see any improvements or disasters coming up.

I test playing in ffplay or converting /w ffmpeg, I get also next error msgs: '[NULL @ 0x7f337002ad00] Skipped PREFIX SEI 6'
Lots of times..  eq. >30 lines per problem notify.

Not every problem notify seems to be related to a Continues counter problem.
.. Maybe is that because all buffering at all different places.

VERSION:
tsversion: TSDuck - The MPEG Transport Stream Toolkit - version 3.31-2722
Built Apr 15 2022 - 15:59:13
Using GCC 7.5.0, C++ std 2011.03
System: Ubuntu (Ubuntu 18.04.6 LTS), on Intel x86-64, 64-bit, little-endian, page size: 4096 bytes
Bitrate: 64-bit floating-point value
Dektec: This version of TSDuck was compiled without Dektec support
Web library: libcurl: 7.58.0, ssl: OpenSSL/1.1.1, libz: 1.2.11
SRT library: libsrt version 1.4.4
RIST library: This version of TSDuck was compiled without RIST support

RUNNING:  
/usr/bin/tsp --monitor --buffer-size-mb 32 -I srt --rcvbuf 16777216 --streamid srt:live.newmedia.zone/001b21ed024b/test-tsp-channel5 --caller 192.168.3.25:8080 -P regulate -O ip --packet-burst 7 --enforce-burst --ttl 5 --force-local-multicast-outgoing --local-port 1234 239.1.4.5:1234

1st Snip:
[mpegts @ 0x7f4ff4000c40] Continuity check failed for pid 7031 expected 15 got 5
[mpegts @ 0x7f4ff4000c40] Continuity check failed for pid 7031 expected 10 got 15
[mpegts @ 0x7f4ff4000c40] Continuity check failed for pid 18 expected 5 got 6
[mpegts @ 0x7f4ff4000c40] Continuity check failed for pid 18 expected 7 got 5
[mpegts @ 0x7f4ff4000c40] Packet corrupt (stream = 1, dts = 8509187126).

2nd Snip:
[mpegts @ 0x7f4ff4000c40] Continuity check failed for pid 7031 expected 13 got 1
[mpegts @ 0x7f4ff4000c40] Continuity check failed for pid 7031 expected 7 got 13
[mpegts @ 0x7f4ff4000c40] Continuity check failed for pid 7031 expected 1 got 7
[mpegts @ 0x7f4ff4000c40] Continuity check failed for pid 18 expected 2 got 4
[mpegts @ 0x7f4ff4000c40] Continuity check failed for pid 18 expected 5 got 2
[mpegts @ 0x7f4ff4000c40] Continuity check failed for pid 18 expected 4 got 5
[mpegts @ 0x7f4ff4000c40] Packet corrupt (stream = 1, dts = 8525973891).

3rd Snip:
[mpegts @ 0x7f3370000c40] Continuity check failed for pid 7011 expected 12 got 5
[mpegts @ 0x7f3370000c40] Continuity check failed for pid 7011 expected 10 got 12
[mpegts @ 0x7f3370000c40] Continuity check failed for pid 7011 expected 1 got 10
[mpegts @ 0x7f3370000c40] Continuity check failed for pid 7011 expected 0 got 1
[mpegts @ 0x7f3370000c40] Continuity check failed for pid 7011 expected 5 got 0
[mpegts @ 0x7f3370000c40] Continuity check failed for pid 7013 expected 4 got 5
[mpegts @ 0x7f3370000c40] Continuity check failed for pid 7013 expected 6 got 4
[mpegts @ 0x7f3370000c40] Continuity check failed for pid 7013 expected 5 got 6
[mpegts @ 0x7f3370000c40] Packet corrupt (stream = 1, dts = 286378676).

I hope you could show me some direction to have look.

lelegard, 2022-04-15:
Hi @dutchboontje 

The packet processing inside `tsp` has no room for changing the order of packets. A packet can be modified or deleted but there is simply no technical feasibility to swap two input packets (unless you write a specific plugin for that, which is not the case here).

If your SRT input is already streamed at playout speed (ie. not reading a file at disk speed on the server), I would recommend to remove the `regulate` plugin. If the input has a highly variable bitrate, the `regulate` plugin will regulate at a more stable bitrate, causing delays, maybe up to the point you miss input packets. In that case, CC issues would be due to missing packets. However, your messages really look like swapped packets.

You may also add `-P continuity` just after the input plugin to check if the packets already have CC issues on input. In which case, you need to investigate on SRT and the server.

dutchboontje, 2022-04-25:
Hi @lelegard 

You were right.  After some testing and changing some SRT upstream I could fix the problems.

--
I am using sls (SRT-LIveServer) as srt gateway.
It seems I had latency put to small.
Besides that I had way to large sndbuf's

After changing both settings a real large amount of CC errors disappaered.

Thanks for to hint.

closed 2022-04-25 by dutchboontje

dutchboontje, 2022-10-11:
Hi Thierry,

Thanks for your suggestions, I will test it out.

Best Regards.