tsduck/tsduck #976: [FEAT] Use custom thread names for plugins in tsp

basicmaster, 2022-04-15:
#### Is your feature request related to a problem?
When `tsp` is used with a larger processing chain, it can be desirable to compare how much CPU power the different plugins (i.e. threads) consume.

Currently the threads of a single `tsp` instance cannot be distinguished (`htop` in tree mode, "Show custom thread names" enabled):

[image]

#### Proposed solution:
On Linux, a custom thread name can be set for a thread. So for example in `ts::tsp::InputExecutor::main()`, `ts::tsp::ProcessorExecutor::main()` and `ts::tsp::OutputExecutor::main()` the following call could be added to set the applicable thread name (requires to `#include <sys/prctl.h>`):

```
prctl(PR_SET_NAME, pluginName().toUTF8().c_str());
```

In the above example, this results in the following `htop` output:
[image]

#### Alternatives:
(none)

#### Additional information:
https://man7.org/linux/man-pages/man2/prctl.2.html

lelegard, 2022-04-16:
Hi @basicmaster

Good idea, thanks.
Done. With equivalence on macOS and Windows as well.

basicmaster, 2022-04-18:
Great, works as expected. Thank you for adding this feature!

closed 2022-04-18 by basicmaster