tsduck/tsduck #980: [Q] Filter SCTE35 Events and pass only what are necessary/ ignore what are not needed

maheshvemuri, 2022-04-22:
I have a feed which has multiple SCTE35 markers with different splice event ids. I need to process or use a specific marker with event-id 6. Do we have this option in tsduck?

My input and output is UDP. I have to filter out and pass only SCTE markers with event-id 6. Please help me in solving this. Thanks

lelegard, 2022-04-23:
Hi @reckless-sergio 

What would you like to do exactly? Extract markers with event id 6 outside the TS for inspection, monitoring, etc? Or remove other SCTE35 markers and only leave those with event id 6 in the SCTE35 PID?

reckless-sergio, 2022-04-23:
Hi @lelegard , I want to remove the other markers and only leave those with event-id 6.

lelegard, 2022-04-25:
I see no immediate solution for such a specialized need with the current plugins. A new simple but specialized plugin would be necessary.

However, I am working on more generic solutions for section filtering which may solve this in the next few days or weeks.

lelegard, 2022-04-26:
Hi @reckless-sergio 

In the very latest push, new options `--section-content` and `--section-mask` have been added to `tstables` and plugins `tables` and `sections`. They let you select sections based on selected bits in their content.

Thus, the following command filters all sections in PID xxx and keeps only SCTE-35 sections with splice_insert() commands with event id 6:
~~~
tsp ... -P sections --pid xxx --section-content FC0000000000000000000000000500000006 --section-mask FF000000000000000000000000FFFFFFFFFF ...
~~~
We select table id 0xFC (splice information table) with command type 0x05 (splice_insert) and event id 0x00000006.

You need version 3.31-2730. Either rebuild or check [nightly builds](https://tsduck.io/download/prerelease/).

reckless-sergio, 2022-05-15:
Hi @lelegard 

I tried the same command you have suggested. Surprisingly, the start is good, but after few seconds(10-20) there is no output.
```
tsp --verbose -I ip 7700 -P sections --pid 1000 --keep --section-content FC0000000000000000000000000500000006
 --section-mask FF000000000000000000000000FFFFFFFFFF -O ip 127.0.0.1:50000
```

Am I missing anything?

lelegard, 2022-05-15:
If there is "no output" (I assume you mean no more sections on PID 1000), this means that there is no more section matching the filter & mask.

reckless-sergio, 2022-10-26:
Hi @lelegard, 
It is working as expected with this feature. But can help me understand how do i build hexa-codes of section-content & section-mask? I have other requirements of filtering/masking scte events based on segmentation_type_id.

Thanks

lelegard, 2022-10-26:
> But can help me understand how do i build hexa-codes of section-content & section-mask?

These filters are typically used by hardware section demux in set-top-boxes. They are extremely generic but need a bit-and-byte understanding of the sections to filter.  So, you need to read the SCTE-35 spec carefully, understand the binary patterns of the sections to filter, where are the various fields to filter (for the section-mask) and what are the values (for the section-content). And then experiment.

reckless-sergio, 2022-10-26:
These are the values I came up with
--section-content FC00000000000000000000000000000000000000000000000000000000000000000000000000000036 --section-mask FF000000000000000000000000FFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFF

I am trying to filter only 36 segmentation_type_id