tsduck/tsduck #981: [Q] using sdt plugin to change service_name or provider_name

dutchboontje, 2022-04-25:
Hi,

I was trying to change service_name and/or provider_name in a certain MPEGTS stream.

But it seems when using spaces, plugin failes to load well.
..and complain /w :  Error: sdt: no parameter allowed, use options only

```
Command used in bash:
tsp  --debug=4 --verbose -I http http://<some url> [-P ...] -P sdt --increment-version --service-id 1 --name "NPO 1" --provider "TEST-Lijntje" -O ip 239.1.4.6:1234

```

```
Decoded by TSP as:
* Error: sdt: no parameter allowed, use options only
* Debug: sdt: ====> tsp -P sdt --increment-version --service-id 1 --name '"NPO' '1"' --provider '"TEST-Lijntje"'
* Debug: error when initializing the plugins

```

In above sample putting a space in service_name failes, but also for provider_name it failes.
By removing any space, command seems succesfully executed.

Also I tried adding escape (\) before the space, does not help.
Double quotes (") vs single (') quotes, does no help

Do you have any suggestion how to add spaces ?

lelegard, 2022-04-25:
Hi @dutchboontje 

You don't run that command from a shell window or script, do you? If run from a bash prompt, any other Unix shell or even the infamous DOS prompt, the command is perfectly correct and the space is correctly embedded in the parameter.

On my side, when I try your command, I get this:
~~~
* Debug: sdt: ====> tsp -P sdt --increment-version --service-id 1 --name 'NPO 1' --provider TEST-Lijntje
~~~

You probably launch that tsp command from some environment where the command line parsing rules are different, or with several levels of encapsulation (in which case the special characters must be escaped several times). In all cases, this is not a TSDuck issue, this is a problem from your command launching environment.

dutchboontje, 2022-05-02:
Hi Thierry,

Thank you, for checking.

I am parsing the argument and starting a tsp-process via sudo.
I will look at that again, again thnx..

closed 2022-05-02 by dutchboontje

dutchboontje, 2022-05-26:
HI,
Just, to let you know.

I dropped all sudo/bash scripts and migrated to the python-bindings.

It works creat, now.

Thanks, again.. for this super tool and all of your time you put into this whole project.