tsduck/tsduck #982: [BUG] Teletext descriptor - missing magazine number

enricovittorini, 2022-04-25:
#### Bug description:

From standard:

```
teletext_descriptor(){ descriptor_tag 8 uimsbf
 descriptor_length 8 uimsbf
 for (i=0;i<N;i++){ ISO_639_language_code 24 bslbf
 teletext_type 5 uimsbf
 teletext_magazine_number 3 uimsbf
 teletext_page_number 8 uimsbf
 } } 
```

PSI plugin - PMT:
```
              {
                "#name": "teletext",
                "language_code": "ita",
                "page_number": 777,
                "teletext_type": 2
              }
```

#### Expected behavior:
magazine number is reported in the plugin output

#### Environment:
 - OS: Windows
 - OS version: 10
 - TSDuck full version: 3.30-2710
 - Installation type:

colinu69, 2022-04-25:
page_number 777 is shorthand for
&nbsp;&nbsp;&nbsp;teletext_magazine_number: 7
&nbsp;&nbsp;&nbsp;teletext_page_number: 0x77

From EN 300 706  (3.1: Definitions):
<pre>
      page number: M Pt Pu, where
            M = magazine (range 1 - 8)
            Pt = page number tens (range 0 - F)
            Pu = page number units (range 0 - F)
</pre>

enricovittorini, 2022-04-26:
@colinu69 or course it is. I was  just reporting that difference to request,  if possible, to have an output as per the standard spec.

enricovittorini, 2022-04-26:
thanks @lelegard ,
could you redirect me on how to modify this on the code and in general how to add a new descriptor if needed?
I would like to recompile to have a more "standard compliant output". thanks for your help!

is there a guide or an example explaining?

thanks
E

lelegard, 2022-04-26:
Hi @enricovittorini 

All descriptors are in `src/libtsduck/dtv/descriptors`. The teletext descriptor is in `tsTeletextDesciptor.h` and `.cpp`. Its XML model (from which the JSON syntax is automatically derived) is in `src/libtsduck/config/tsduck.tables.model.xml` (look for `<teletext_descriptor>`).

If you want to add a new descriptor in the TSDuck source code, simply add the corresponding `.h` and `.cpp` files in `src/libtsduck/dtv/descriptors` and its XML model in `src/libtsduck/config/tsduck.tables.model.xml`. If you need to add names for private descriptors, add them in `src/libtsduck/config/tsduck.names`.

Generally speaking, if you need to add some non-confidential descriptor, I recommend to update the TSDuck source code and submit a pull request for insertion in the mainstream code ([see here](https://tsduck.io/doxy/contributing.html)).

If you want to add confidential information, I would recommend to keep it outside the main source tree to avoid resynchronizing each time you pull a new version. Instead, write you own "extension". See an example in `sample/sample-extension`. The sample extension adds a table, a descriptor, a section filter and a CAS. You probably don't need all of them. The sample scripts build Linux and Windows installers. Thus, you create a completely private extension that you install on top of TSDuck. Just be sure to rebuild and reinstall your extension when you install a new TSDuck version.