tsduck/tsduck #984: [BUG] Exception at tsproc.start() when sample-app is run

satyamsameer, 2022-04-29:
#### Bug description:
Exception at tsproc.start() when sample-app is run

#### How to reproduce:
1. Copy sample application from sample directory
2. Apply the tsduck.props in project provided in tsduck x64 setup
3. Run in debug mode

#### Expected behavior:
Code should dump .ts output without any exception

#### Errors and logs:
[image]

#### Environment:
 - OS: Windows 10
 - OS version: Version 1903 (OS Build 18362.175)
 - TSDuck full version: 3.30-2710
 - Installation type: x64 with C++ development option

#### Additional information:
Using VS 2019 with Windows SDK 10 and toolset v142

lelegard, 2022-05-01:
Hi @satyamsameer 

This is one of the usual weirdnesses from Windows: DLL's are not compatible between debug and release versions. You have the problem on the debug version (seen on your screenshot) and I would bet you do not have it with the release version.

As far as I can understand Windows DLL hell, the situation is the following:
- Application is compiled in debug mode
- Linked with debug version of `tsduck.dll` (from `<Import Project="$(TSDUCK)\tsduck.props"/>` in the application project file).
- At runtime, the plugin `until` is invoked and the DLL `tsplugin_until.dll` is activated.
- This DLL is located in the installation tree of TSDuck and is a release version. Code in that DLL was compiled and linked with a release version of `tsduck.dll`. However, when this code is executed in the context of a debug version of `tsduck.dll`. The two versions (debug vs. release) of the _same_ DLL, built from the _same source code_ are not compatible, hence the crash. Thank you Mr. Windows....

I can reproduce the problem with the same configuration. Then, I copied a debug version of `tsplugin_until.dll` in the sample app debug output directory and it worked.

You wouldn't have that kind of problem with "civilised" operating systems such as Linux or macOS where the debug and release versions of the same code are interoperable.

As an additional note, I noticed that the sample applications missed an initialisation step which is required on Windows only. As _yet another weirdness_, Windows applications cannot run when networking and COM have not been explicitly initialised. This has been fixed in the sample applications code.

satyamsameer, 2022-05-02:
Thanks @lelegard for the reply

I am now running sample-app in release build with the latest changes as you mentioned and it is now generating output.ts perfectly with tsduck.dll release library and release plugins as well.

Is there any chance you can provide plugins dll build with debug tsduck.dll so that I can debug my code. I require this to integrate tsduck library with my other projects as they compile in all configurations and platforms on windows.

Or I have to do it myself from source code but it will take time as I don't have any prior experience regarding this.

Thanks

lelegard, 2022-05-02:
>Is there any chance you can provide plugins dll build with debug tsduck.dll 

Still possible but I would need to provide all configurations, 33 and 64 bits, debug and releases. The Windows installer is already huge (more than 60 MB) while it is 10 times smaller on Linux.

> Or I have to do it myself from source code but it will take time as I don't have any prior experience regarding this.

This is [fully documented here](https://tsduck.io/doxy/building.html). Since you are already a C++ developer on Windows with Visual Studio installed, most of the difficulty is already over. Just run the PowerShell script `scripts\install-prerequisites.ps1` once to install other prerequisites, then the script `scripts\build.ps1` builds everything.