tsduck/tsduck #986: [BUG] analyze plugin:  json-udp - datagram larger than expected

enricovittorini, 2022-04-30:
#### Bug description:
Trying to send the output of tsanalyze in json format over UDP. on a large TS the following error is raised:
 Error: analyze: error sending UDP message: A message sent on a datagram socket was larger than the internal message buffer or some other network limit, or the buffer used to receive a datagram into was smaller than the datagram itself

#### Errors and logs:

`Error: analyze: error sending UDP message: A message sent on a datagram socket was larger than the internal message buffer or some other network limit, or the buffer used to receive a datagram into was smaller than the datagram itself`
#### Environment:
 - OS: Windows
 - OS version: 10
 - TSDuck full version: 3.31-2730
 - Installation type: official

lelegard, 2022-04-30:
The hard-coded limit of an IP datagram is 64 kB. If the JSON data is larger, there is no way to send it in one piece using UDP. As a test, you may try option `--json-line` and check the resultant size (same format as `--json-udp`). Don't use `--json` since the spaces for indentation would inflate the JSON size. If the size is significantly smaller than 64 kB, this may be a buffer size issue and an additional option to set the socket buffer size would solve this (or always arbitrarily set the UDP buffer size for that socket to 64 kB).

> split the data in multiple packets and use TCP

In fact, this is rather "or" than "and". In the case of TCP, this is a stream.

It is possible to use TCP. However, this would significantly complicate the plugin. The plugin is part of a chain. Although the multi-threaded nature of tsp (one thread per plugin) mitigates the risk of blocking the chain when one plugin blocks, there is a general rule that a plugin shall not put itself in a position where it could block more than a few milliseconds. Sending a UDP packet is instantaneous, writing a short text file is very fast. Connecting to a TCP socket, waiting for the server, full exchange and socket shutdown is usually fast but we have all experienced sporadic latencies of several seconds in TCP exchanges. So, I do not like the idea of a full TCP session in the middle of a packet processing chain. In that case, I usually create another thread, enqueue the data to send, resynchronise on thread termination, etc. This is typically what the `scrambler` plugin does when connecting to an ECMG using a TCP session.

There is an additional complexity: The command `tsanalyze` and the plugin `analyze` share the same code base, of course. The execution context is very different, one standalone command without addition constraint vs. a multi-threaded application. The common code is simply a class with synchronous non-blocking (or short-blocking) methods. The integration of a TCP in a common code with ancillary threads for heterogeneous contexts also adds some complexity.

Nothing impossible, just more work. That's why starting with the JSON size evaluation and tweak the UDP buffer size would be faster.

enricovittorini, 2022-04-30:
thanks @lelegard 
 
the json-line is 572KB, definitely bigger than the 64KB.
What would be the solution here in your opinion? a UDP delivery is a guess that the 64KB are not exceeded.

lelegard, 2022-05-01:
Hi @enricovittorini 

I added option `--json-tcp` to report the JSON text through a TCP connection. By default, a new TCP connection is initiated for each message (e.g. with `-P analyze --interval`). With option `--json-tcp-keep`, the same TCP connection is kept open for all messages.

Just in case, I also added option `--json-buffer-size` which sets the socket buffer size for TCP and UDP.

I chose to be simple, even though not ideal in some cases. The TCP communication is synchronous. This can introduce some latency with `-P analyze --interval`. This is documented.

The same set of JSON options is automatically inherited by all commands and plugins which use JSON output.

enricovittorini, 2022-05-02:
Thanks @lelegard , will test it and let you know. Just one question: whe TCP connection is kept during the session I guess there is the need to specify a "delimiter" to be used on the server side when the data is fully received. 

Do you agree?
Thanks
E

lelegard, 2022-05-02:
>  whe TCP connection is kept during the session I guess there is the need to specify a "delimiter" to be used on the server side when the data is fully received.

The JSON structure is sent as one line in compact form (same as JSON log line or UDP message). The end of line is sent as CR/LF (telnet protocol format). So, either the server reads one line of text and then parses it as JSON, or the server performs a direct JSON parsing and the closing bracket matching the initial opening one naturally signals the end of structure.

On the other direction, the server is not expected to send anything, this is really a one way communication.

enricovittorini, 2022-05-02:
@lelegard 
tested and it seems working well. thanks a lot.
The issue I see so far is that if the json-tcp-keep is used and the server is restarted, the plugin does not try to reconnect and show the following error:
`Error: analyze: error sending data to socket: An existing connection was forcibly closed by the remote host`

It would be good to have automatic reconnection and a configurable timeout possibly.
I was testing the "-i" report and in case of server crash and restart this is a problem.

any plan to add this functionality to other plugin like tables, psi, bitrate_monitor?

lelegard, 2022-05-03:
> It would be good to have automatic reconnection

Done

enricovittorini, 2022-05-04:
> 

@lelegard  tested build 3.31-2737
when tcp-keep is used in analyze plugin  bitrate goes to down to 0 after a some seconds.

*```
 Debug[2]: ip: received UDP packet, source: 20.0.0.100:54434, destination: 239.0.0.1:5001, timestamp: -1
* Debug[2]: ip: received UDP packet, source: 20.0.0.100:54434, destination: 239.0.0.1:5001, timestamp: -1
* Debug[2]: ip: received UDP packet, source: 20.0.0.100:54434, destination: 239.0.0.1:5001, timestamp: -1
* bitrate_monitor: 2022/05/04 12:02:41, BTMN: TS bitrate: 5,275,280 bits/s
* Warning: bitrate_monitor: BTMN: TS bitrate (0 bits/s) is lower than allowed minimum (10 bits/s)
* bitrate_monitor: 2022/05/04 12:02:43, BTMN: TS bitrate: 0 bits/s
* bitrate_monitor: 2022/05/04 12:02:45, BTMN: TS bitrate: 0 bits/s
* bitrate_monitor: 2022/05/04 12:02:47, BTMN: TS bitrate: 0 bits/s
* [MON] 2022/05/04 12:02, VM: 33 MB (stabilizing), CPU:4.37% (average:61.33%)
* bitrate_monitor: 2022/05/04 12:02:49, BTMN: TS bitrate: 0 bits/s

```

without the tcp-keep option it seems ok.

lelegard, 2022-05-04:
> when tcp-keep is used in analyze plugin bitrate goes to down to 0 after a some seconds.

This means that the `tsp` stream is blocked, packets no longer fly across the pipeline of plugins (ie. TS bitrate = 0).

This is probably exactly what I originally predicted and my reluctance to use TCP (synchronous stream, wait for peer) instead of UDP (asynchronous message, fire and forget).

The most probable cause is that the TCP server is not reading its client socket. When the TCP windows and the send and receive buffers are full, the client process (here `tsp` in the `analyze` plugin thread) is blocked, waiting for the peer to read data and clear space in the TCP flow control.

Without `--tcp-keep`, a new connection is opened / closed each time. If your server is not responsive, the connection fails and the flow continues. With `--tcp-keep`, the connection is already established and you wait for the peer.

By using TCP, your `tsp` flow becomes a slave of the TCP server. You wanted TCP, you got it....