tsduck/tsduck #987: [BUG] 'pcrextract' plugin stops reporting the PTS when the PTS is looping back

moving-digital, 2022-04-30:
Hello Thierry @lelegard 

#### Bug description:
The `pcrextract` plugin stops reporting the PTS when the PTS is looping back, regardless if the input is a file or live stream.
The command I'm using is:  `-P pcrextract --noheader --pts --good-pts-only --pid 301`

#### How to reproduce:
The two following use-cases will allow to reproduce the bug:
1. Get a TS file, play it out in input of TSP with the `--infinite` option, then wait for the file to loop back to the start.
2. Use a Live stream (DVB Tuner), play it out in input of TSP, then wait for the PTS to reach its maximum value: 8589934591.

#### Expected behavior:
1. For use-case 1: `TSP` or `pcrextract` plugin should be aware that the file has looped, so that the new PTS that actually has a lower value than the previous good PTS, is still a good PTS.
2. For use-case 2: `TSP` or `pcrextract` plugin should be aware that the live PTS has looped back from 8589934591 to 0, so that the new PTS that actually has a lower value than the previous good PTS, is still a good PTS.

#### Errors and logs:
No error, no logs. Just no PTS value displayed on the stdout.

#### Environment:
 - OS: Linux
 - OS version: Debian 10
 - TSDuck full version: 3.31-2730
 - Installation type: Official binary

lelegard, 2022-04-30:
> TSP or pcrextract plugin should be aware that the file has looped, so that the new PTS that actually has a lower value than the previous good PTS, is still a good PTS.

That is impossible by design. All plugins are fully independent from each other. The input plugin is just a packet source. The fact that it is a file and that it loops back is unknown to tsp and other plugins. When the input plugin generates a leap in PCR or CC or whatever, this is just an anomaly. 

That being said, I will investigate what could be a reasonable behaviour in `pcrextract`.

moving-digital, 2022-05-01:
Thanks a lot, Thierry!

lelegard, 2022-05-01:
Hi @moving-digital 

I originally did not pay attention that you used option `--good-pts-only`.

As documented:
~~~
  --good-pts-only
      Keep only "good" PTS, ie. PTS which have a higher value than the previous
      good PTS. This eliminates PTS from out-of-sequence B-frames.
~~~
This means that what you are asking for is impossible by design. When the file loops back, the PTS leap backward and will never be higher than the last "good PTS", which was the last goof PTS before the end of file.

So, this is not a bug, this is the intended behaviour.

Without option `--good-pts-only`, PTS continue to be reported when the file loops back.

moving-digital, 2022-05-01:
Hi @lelegard 

Yes, I'm using --good-pts-only but using that option with a live stream still works when the PTS rolls back to 0.

However it is not the case with a file as input. 

Is that a normal behavior?

lelegard, 2022-05-01:
> Yes, I'm using --good-pts-only but using that option with a live stream still works when the PTS rolls back to 0.

It depends on what you mean by "roll back".

In TSDuck, we consider modular integer roll back on PCR/PTS/DTS as "normal". However, looping back from some arbitrary value to another lower arbitrary value as an error.

This is similar to continuity counters. CC=15 followed by CC=0 is normal, just a modular arithmetic roll back. On the other hand, CC=12 followed by CC=7 is a discontinuity.

moving-digital, 2022-05-01:
Hi @lelegard 

What I mean by "PTS roll back" in a live stream scenario is when the PTS value leaps from its max value = 8589934591 back to 0.

In this case, even when using --good-pts-only works and "pcrextract" plugin will not get stuck at 8589934591. Instead, it will start reporting the PTS value again from 0 onwards.

lelegard, 2022-05-01:
> In this case, even when using --good-pts-only works and "pcrextract" plugin will not get stuck at 8589934591. Instead, it will start reporting the PTS value again from 0 onwards.

That is exactly what I explained in the previous post. This is a "normal" continuity of PTS values. PTS values continue to be good after modular integer wrap-up. On the other hand, after an arbitrary roll back, such as a file looping back to the beginning, there is no continuity in the PTS values, which are consequently no longer considered as "good".

That is why the observed behaviour is correct and expected with `--good-pts-only`.

moving-digital, 2022-05-02:
Noted with thanks, Thierry!

closed 2022-05-02 by moving-digital