tsduck/tsduck #988: [Q] analyze plugin behavior check

enricovittorini, 2022-05-03:
Hello @lelegard I would have two questions on the analyze plugin:

1) I am analyzing a TS every 2 seconds with the analyze plugin and from time to time I get the "services" in the json output  all with the name as "unknown" and a couple with name = System Software Update.  this is strange as SDT at the same time seems ok.

```
		{
			"bitrate": 2264165,
			"bitrate-204": 2456860,
			"components": {
				"clear": 3,
				"scrambled": 2,
				"total": 5
			},
			"id": 8146,
			"is-scrambled": true,
			"name": "(unknown)",
			"original-network-id": 0,
			"packets": 3012,
			"pcr-pid": 173,
			"pids": [
				173,
				420,
				1016,
				1816,
				5507
			],
			"pmt-pid": 1016,
			"provider": "(unknown)",
			"ssu": false,
			"t2mi": false,
			"tsid": 6300,
			"type": 0,
			"type-name": "unknown (0x00)"
		},
		{
			"bitrate": 1634980,
			"bitrate-204": 1774127,
			"components": {
				"clear": 5,
				"scrambled": 0,
				"total": 5
			},
			"id": 8173,
			"is-scrambled": false,
			"name": "(System Software Update)",
			"original-network-id": 0,
			"packets": 2175,
			"pids": [
				86,
				97,
				102,
				106,
				111
			],
			"pmt-pid": 86,
			"provider": "(unknown)",
			"ssu": true,
			"t2mi": false,
			"tsid": 6300,
			"type": 0,
			"type-name": "unknown (0x00)"
		}
```

Second question is about the bitrate evaluation of the plugin: it seems that, when the "-i" option is specified, the repetition-ms of the tables is evaluated only for the interval duration. So if for example the interval is 2s and the tables has a repetition rate of 5s, then you get a repetition-ms of 0Kbps. is this confirmed?

thnaks!

enricovittorini, 2022-05-03:
to be closed. issue i think is linked to a variable SDT rep-rate. if analysis is not complete in the interval there is no way to get the service name.

closed 2022-05-03 by enricovittorini

lelegard, 2022-05-03:
Hi @enricovittorini 

The analysis is restarted from scratch after each interval. Intervals smaller than 20 seconds do not really make sense since some tables may be completely missed (eg. TDT/TOT).

enricovittorini, 2022-05-03:
@lelegard agree.
I was trying to have a real-time update of the repetition rate of the tables. maybe a possible solution would be concatenate multiple filter and analyze with increasing interval.....

`-P analzye -i 1 ... -P filter --psi-si .. -P analyze -i 2 -P filter -n  17 -P analyze -i 10 etc... `

one more thing as we talk about analyze plugin...
I was looking for a way to have the whole TS pids bitrate in almost real-time (every 2 sec max ). I only found analyze to report them. Something else I should try?

lelegard, 2022-05-03:
As a general consideration, measuring "intervals" (bits/second or tables/second) in "real time" is a non-sense since you need a measurement duration which is larger than the interval to be measured, thus no longer being "real time". Two or three times the measured interval is even preferable.

> I was looking for a way to have the whole TS pids bitrate in almost real-time (every 2 sec max ). 

Plugin `analyze` with 2 seconds intervals is fine to evaluate the bitrate, not the table intervals, at least tables with large cycles.