tsduck/tsduck #989: [BUG]

lmoody1, 2022-05-03:
#### Bug description:
Output stream doesn't start if a null input stream is supplied

#### How to reproduce:
        tsp --control-local localhost --control-port 8000 -v -b 41000000 -I null -j -O ip localhost:8999 
        -P merge -j --no-psi-merge --ignore-conflicts -p 0x1-0x20 -d 0x0 -d 0x21-0x1FFE "tsp -v -b 3000000 -I file -i C:/Files/Stream_MULx21.ts -P regulate" 
        -P regulate 
        -P merge -j -t "tsp -b 2000000 -v -I null -P regulate" 
        -P pat 
        -P pmt -c 
        -P regulate
        
#### Expected behavior:
The merged streams (except for the null stream) should output to localhost: 8999 and be seen on (for example) tsReader. Expecting to see (at the minimum) "TOT", "SDT", "NIT", "CAT" and "TDT" PIDs. Not expected to see PAT. See "**Merged Streams without Null Input Stream (Works as Expected)**" in the attached document as an example of what we should be seeing.

#### Errors and logs:
The merged stream isn't being output. See "**Merged Streams with Null Input Stream (Not Working as Expected)**" in attached document.

#### Environment:
 - OS: Windows 
 - OS version: Windows 10
 - TSDuck full version: 3.30-2710
 - Installation type: official binary TSDuck-Win64-3.30-2710.exe

#### Additional information:
Output is as expected if the null input stream is restarted with new parameters (see **Merged Streams with Null Input Stream Restarted with New Parameters  (Works as Expected)** in attached document):

        tspcontrol -t localhost:8000 restart -v 3 -j -t "tsp -b 2000000 -v -I file -i C:/Files/Stream_8E1.ts -P regulate" 
        tspcontrol -t localhost:8000 restart -v 4 -c -a 0x8FF6/0x21 
        tspcontrol -t localhost:8000 restart -v 5 -a 0x8E1/0x5 -s 0x8FF6 --pcr-pid 0x1fff 

Output is as expected if the stream is executed without the null input stream  (**see Merged Streams without Null Input Stream (Works as Expected** in attached document):

        tsp --control-local localhost --control-port 8000 -v -b 41000000 -I null -j -O ip localhost:8999 
        -P merge -j --no-psi-merge --ignore-conflicts -p 0x1-0x20 -d 0x0 -d 0x21-0x1FFE "tsp -v -b 3000000 -I file -i C:/Files/Stream_MULx21.ts -P regulate" 
        -P regulate 

[Merged Streams with Null Input Stream Bug.docx](https://github.com/tsduck/tsduck/files/8613968/Merged.Streams.with.Null.Input.Stream.Bug.docx)

lelegard, 2022-05-04:
Hi @lmoody1 

I do not know what is the exact issue here but I would recommend to start with a more rational usage of the `regulate` plugin. There is no need to put two of them in sequence, it is at best useless and at worst counterproductive. Moreover, the first `merge` is placed before the first `regulate`, meaning that the secondary stream is merged on an unregulated stream of null packets.

Additionally, there is no need for `-j` options if you do not use "joint termination".

I have no idea if this will fix the issue but you should start with a more rational regulation, something like:
~~~
tsp -b xxxxx ....\
    -I null \
    -P regulate \
    -P merge "tsp -I file ... -P regulate" \
    -P merge "tsp -I file ... -P regulate" \
    -P ...
    -O ip ...
~~~

lmoody1, 2022-05-04:
I've tried this with a multitude of variations of locations of the regulate plugin(s), as well as the addition, removal and ordering of pat & pmt plugins. I have also tried it without the -j option. I have not been able to get it to work.

My goal is to be able to start and stop input streams at any time with new stream settings. I'm doing this by creating a muxed output stream with a file input stream merged with multiple null input streams that can be modified and restarted at any time with the users' requested configurations. This is the basic logical:

1. input from a file
2. input from null stream
3. input from null stream
4. input from null stream
5. output to localhost

In the above example, null input streams 2, 3, 4 are available to the users to restart at **_any time_** with user requested settings (i.e. input from file "c:\files\xxx.ts", with a bitrate of "2000000", to service "0x8FF1", and input PID "0x8e1" mapped to PID "0x8E2")

What's happening is:

- If I create a merge stream with **_only_** the input from file stream (i.e. 1 above **_without_** 2, 3, or 4) I **_am_** seeing the output.  
- If I create a merge stream with the input from a file merged with the null streams (i.e. 1 above merged with 2, 3 and 4). I am _**not**_ seeing the output (I don't think it's executing the stream at all)
- If I create a merge stream with the input from a file merged with a null stream (i.e. 1 above merged with 2, 3 and 4) I am _**not**_ seeing the output, but then I _**restart**_ one of the null input streams with user requested settings (i.e. restart 2 above with input from file "c:\files\xxx.ts", with a bitrate "2000000", to service "0x8FF1", and input PID "0x8e1" mapped to PID "0x8E2"), I **_am_** seeing the output.

One thing that **_does_** work is if I specify a service and PID for the null input streams. I.E. with a service starting at 0xA0 and PID starting at 0xB0:

        tsp --control-local localhost --control-port 8000 -v -b 41000000 -I null -j -O ip localhost:8999 
        -P merge -j --no-psi-merge --ignore-conflicts -p 0x1-0x20 -d 0x0 -d 0x21-0x1FFE "tsp -v -b 3000000 -I file -i C:/files/input_stream.ts -P regulate" 
        -P regulate 
        -P merge -j -t "tsp -b 100000 -v -I null -P regulate" 
        -P pat -c -a 0xA0/0x21 
        -P pmt -c -a 0xB0/0x5 -s 0xA0 --pcr-pid 0x1fff 
        -P merge -j -t "tsp -b 100000 -v -I null -P regulate" 
        -P pat -c -a 0xA1/0x21 
        -P pmt -c -a 0xB1/0x5 -s 0xA1 --pcr-pid 0x1fff 

However, this requires us to create a dummy service and a dummy PID for each null input stream. We'd prefer not to have to do this. It would require us to reserve those values so the users won't be able to use them (and we're looking to reserve 32 null streams so that will eliminate 32 services and PIDs for users to use). Instead, we'd prefer to do something like this:

        tsp --control-local localhost --control-port 8000 -v -b 41000000 -I null -j -O ip localhost:8999 
        -P merge -j --no-psi-merge --ignore-conflicts -p 0x1-0x20 -d 0x0 -d 0x21-0x1FFE "tsp -v -b 3000000 -I file -i C:/files/input_stream.ts -P regulate" 
        -P regulate 
        -P merge -j -t "tsp -b 100000 -v -I null -P regulate" 
        -P pat 
        -P pmt -c 
        -P merge -j -t "tsp -b 100000 -v -I null -P regulate" 
        -P pat 
        -P pmt -c 

I hope this all makes sense.

I appreciate your help.

-Lee

closed 2022-06-02 by lmoody1