tsduck/tsduck #990: Is it possible to build tsduck using conda environments?

kkhatr2, 2022-05-03:
I have access to a RHEL system without without `sudo`. So, I cannot install from prebuilt binaries or execute `install_prerequisites.sh` 

I do have access to `conda` and have created an environment with appropriate libraries and packages that I sniffed out of the `install_prerequisites.sh` file.

With the environment created. I execute start of compilation with, `make -j30 NODEKTEC=1 NOHIDES=1 NOCURL=1 NOPCSC=1 NOEDITLINE=1 NOSRT=1 NORIST=1 NOTELETEXT=1`,  the compilation starts and [DEP] completes, but [CXX] starts and exits with the stacktrace below.

So, the question is, is it possible to compile tsduck without `sudo` access and, with a package manager like `conda` or some other method?

```
In file included from dtv/broadcast/tsModulation.h:37,
                 from dtv/broadcast/tsModulationArgs.h:40,
                 from dtv/broadcast/tsChannelFile.h:36,
                 from dtv/broadcast/tsChannelFile.cpp:30:
dtv/broadcast/tsDeliverySystem.h:52:30: error: '::SYS_TURBO' has not been declared
   52 |         DS_DVB_S_TURBO   = ::SYS_TURBO,
      |                              ^~~~~~~~~
bash: warning: setlocale: LC_ALL: cannot change locale (en_US.UTF-8)
  [CXX] dektec/tsDektecControl.cpp
dtv/broadcast/tsDeliverySystem.h:54:30: error: '::SYS_DVBT2' has not been declared; did you mean SYS_DVBT'?
   54 |         DS_DVB_T2        = ::SYS_DVBT2,
      |                              ^~~~~~~~~
      |                              SYS_DVBT
In file included from dtv/broadcast/tsModulation.h:37,
                 from dtv/broadcast/tsModulationArgs.h:40,
                 from dtv/broadcast/tsBitrateDifferenceDVBT.h:36,
                 from dtv/broadcast/tsBitrateDifferenceDVBT.cpp:30:
dtv/broadcast/tsDeliverySystem.h:52:30: error: '::SYS_TURBO' has not been declared
   52 |         DS_DVB_S_TURBO   = ::SYS_TURBO,
      |                              ^~~~~~~~~
dtv/broadcast/tsDeliverySystem.h:54:30: error: '::SYS_DVBT2' has not been declared; did you mean SYS_DVBT'?
   54 |         DS_DVB_T2        = ::SYS_DVBT2,
      |                              ^~~~~~~~~
      |                              SYS_DVBT
In file included from dtv/broadcast/tsModulationArgs.h:40,
                 from dtv/broadcast/tsChannelFile.h:36,
                 from dtv/broadcast/tsChannelFile.cpp:30:
dtv/broadcast/tsModulation.h:365:22: error: '::TRANSMISSION_MODE_1K' has not been declared; did you mean 'TRANSMISSION_MODE_8K'?
  365 |         TM_1K    = ::TRANSMISSION_MODE_1K,
      |                      ^~~~~~~~~~~~~~~~~~~~
      |                      TRANSMISSION_MODE_8K
dtv/broadcast/tsModulation.h:366:22: error: '::TRANSMISSION_MODE_16K' has not been declared; did you mean 'TRANSMISSION_MODE_8K'?
  366 |         TM_16K   = ::TRANSMISSION_MODE_16K,
      |                      ^~~~~~~~~~~~~~~~~~~~~
      |                      TRANSMISSION_MODE_8K
In file included from dtv/broadcast/tsModulationArgs.h:40,
                 from dtv/broadcast/tsBitrateDifferenceDVBT.h:36,
                 from dtv/broadcast/tsBitrateDifferenceDVBT.cpp:30:
dtv/broadcast/tsModulation.h:365:22: error: '::TRANSMISSION_MODE_1K' has not been declared; did you mean 'TRANSMISSION_MODE_8K'?
  365 |         TM_1K    = ::TRANSMISSION_MODE_1K,
      |                      ^~~~~~~~~~~~~~~~~~~~
      |                      TRANSMISSION_MODE_8K
dtv/broadcast/tsModulation.h:367:22: error: '::TRANSMISSION_MODE_32K' has not been declared; did you mean 'TRANSMISSION_MODE_2K'?
  367 |         TM_32K   = ::TRANSMISSION_MODE_32K,
      |                      ^~~~~~~~~~~~~~~~~~~~~
      |                      TRANSMISSION_MODE_2K
dtv/broadcast/tsModulation.h:366:22: error: '::TRANSMISSION_MODE_16K' has not been declared; did you mean 'TRANSMISSION_MODE_8K'?
  366 |         TM_16K   = ::TRANSMISSION_MODE_16K,
      |                      ^~~~~~~~~~~~~~~~~~~~~
      |                      TRANSMISSION_MODE_8K
dtv/broadcast/tsModulation.h:367:22: error: '::TRANSMISSION_MODE_32K' has not been declared; did you mean 'TRANSMISSION_MODE_2K'?
  367 |         TM_32K   = ::TRANSMISSION_MODE_32K,
      |                      ^~~~~~~~~~~~~~~~~~~~~
      |                      TRANSMISSION_MODE_2K
dtv/broadcast/tsModulation.h:417:26: error: '::GUARD_INTERVAL_1_128' has not been declared; did you mean 'GUARD_INTERVAL_1_8'?
  417 |         GUARD_1_128  = ::GUARD_INTERVAL_1_128,
      |                          ^~~~~~~~~~~~~~~~~~~~
      |                          GUARD_INTERVAL_1_8
dtv/broadcast/tsModulation.h:417:26: error: '::GUARD_INTERVAL_1_128' has not been declared; did you mean 'GUARD_INTERVAL_1_8'?
  417 |         GUARD_1_128  = ::GUARD_INTERVAL_1_128,
      |                          ^~~~~~~~~~~~~~~~~~~~
[... 121 lines cut ...]
```

My env.yml file looks like below.

```
name: compile
channels:
    - conda-forge
dependencies:
    - dos2unix
    - curl
    - zip
    - doxygen
    - graphviz
    - libedit
    - libcurl
    - rpm-tools
    - python
    - openjdk
    - gxx
    - gcc
    - make
    - perl
    - binutils
```

lelegard, 2022-05-04:
Hi @kkhatr2 

I do not know the conda environment. However, it is absolutely possible (and even recommended) to build TSDuck without root privilege. The installation can be performed anywhere using `make install SYSPREFIX=/any/where`. This can be an existing or empty system tree. Just add the `bin` subdirectory in your `PATH`. [See more details here](https://tsduck.io/doxy/building.html).

As an example, TSDuck is already part of the LinuxBrew environment, a user-only package management for Linux, maybe something similar to conda.

The script `install_prerequisites.sh` is just a fast convenience to setup the build prerequisites (for instance in a CI/CD pipeline), nothing more. Just read it to check the required packages but you have already realized that.

All compilation errors you get are due to the lack of system header file `/usr/include/linux/dvb/frontend.h`. It is [included here](https://github.com/tsduck/tsduck/blob/4cbe15772a33b1686449c172db08f99cc777e071/src/libtsduck/base/types/tsPlatform.h#L883) in TSDuck source code. You should probably have had a compilation here.

In RHEL, this header file comes with the package `kernel-headers`. I realize that this dependency is missing from `install_prerequisites.sh`. I will add it (as well as the equivalent for all other distros). If you install the corresponding conda package, it should be fine.

lelegard, 2022-05-05:
Dektec support should not fail. The Dektec DTAPI is automatically downloaded by the makefile (requires an Internet connection) and has no other dependency.

About libedit and libcurl, I do not know. This must be a conda specific issue. I have no idea how this is organized.

As I already explained, TSDuck works in the LinuxBrew environment, a local user-only installation of all Linux tools. So, having libedit and libcurl in such an environment is not a problem. You should investigate on the conda side, I cannot help.

kkhatr2, 2022-05-05:
Hello @lelegard 

I could get tsduck installed with LinuxBrew in my local environment as well.

I do appreciate your guidance in identifying the missing dependency of kernel-headers to build from sources.  
Now that I can successfully compile. I can work on peeling back the layers to build tsduck with all features using conda environment.

closed 2022-05-05 by kkhatr2