tsduck/tsduck #991: [BUG]

enricovittorini, 2022-05-05:
#### Bug description:
In analyze plugin  the PID description for teletext report "Teletext (ita, Initial Teletext page)" while teletext contains more than one language.
description on this case I think should be "private data"

PID in tables.pids:

```
{
      "audio": false,
      "bitrate": 37631,
      "bitrate-204": 40834,
      "description": "Teletext (ita, Initial Teletext page)",
      "ecm": false,
      "emm": false,
      "global": false,
      "id": 712,
      "invalid-pes-prefix": 0,
      "is-scrambled": false,
      "language": "ita",
      "packets": {
        "af": 0,
        "clear": 251,
        "discontinuities": 0,
        "duplicated": 0,
        "invalid-scrambling": 0,
        "pcr": 0,
        "scrambled": 0,
        "total": 251
      },
      "pes": 251,
      "pes-stream-id": 189,
      "pmt": false,
      "service-count": 1,
      "services": [
        7779
      ],
      "t2mi": false,
      "unreferenced": false,
      "video": false
    }
```

the part of interest of the PMT for  this service:

```
{
			"#name": "component",
			"#nodes": [
				{
					"#name": "teletext_descriptor",
					"#nodes": [
						{
							"#name": "teletext",
							"language_code": "ita",
							"page_number": 100,
							"teletext_type": 1
						},
						{
							"#name": "teletext",
							"language_code": "ita",
							"page_number": 777,
							"teletext_type": 2
						},
						{
							"#name": "teletext",
							"language_code": "eng",
							"page_number": 778,
							"teletext_type": 2
						}
					]
				}
			],
			"elementary_pid": 712,
			"stream_type": 6
		}
```

#### Expected behavior:
description on this case I think should be "private data"

#### Environment:
 - OS:  Windows
 - OS version: 10
 - TSDuck 3.31-2737
 - Installation type: official

lelegard, 2022-05-05:
HI @enricovittorini 

What exactly should be "private data"?

The XML definition of the teletext descriptor is given in section D.8.114 of the TSDuck user's guide. It is inlined with the DVB description in ETSI EN 300 468  (6.2.43). The JSON structure is produced by automatic XML-to-JSON translation as defined in 2.7.3 of the TSDuck user's guide.

enricovittorini, 2022-05-05:
@lelegard  i mean the "description" field of the json. 
it reports the initial teletext page: only part of the teletext descriptor in the PMT. --> "Teletext (ita, Initial Teletext page)".

being a stream_type = 6, I think a more correct "description" in the json for the PID (the analzye.pids is about "pids") should be "private data"

lelegard, 2022-05-05:
> being a stream_type = 6, I think a more correct "description" in the json for the PID (the analzye.pids is about "pids") should be "private data"

OK, I see. Well, the analyze plugin is here to give a clear view of what is in the stream. A human reader is more interested to know that a PID carries Italian teletext. The technical description "private data" is completely useless. This is done on purpose.

enricovittorini, 2022-05-05:
@lelegard  but as you can see in that PIDs there are two teletext ita pages and one eng. There would be missing info for the users. if you prefer to keep that level of info, maybe better to have only "teletext"

lelegard, 2022-05-08:
OK, the multiple languages handling in the analysis was a bug. Just fixed it.

In the JSON analysis report, the "language" field remains a string with the first language code, for compatibility. There is now an additional "languages" field (with an 's') which is an array of strings, all languages in the PID.