tsduck/tsduck #993: [Q] EIT XML file sample

Litrona, 2022-05-06:
Where can I find a real sample of EIT content XML file, as described in page 471 of user's manual?
I mean with the descriptor list contents, that I guess it will contain the event name and the text to show.

Thanks in advance.
Regards

lelegard, 2022-05-06:
Hi @Litrona 

The easiest way is to extract EIT's from some existing stream. If you do not have any, pick up one from the [TSDuck stream repository](https://tsduck.io/streams/), for instance:
~~~
tsp -I http https://tsduck.io/streams/uk-italy-bskyb/sky_282E_10714.ts \
    -P tables --pid 18 --all-once --pack-all-sections --xml eit.xml --max-tables 10 \
    -O drop
~~~
The above command extracts the first 10 EIT's from the designated file.

Litrona, 2022-05-08:
I have an SPTS with PAT, PMT and SDT tables and lots of null packets (above 1 Mbps). I want to add an EPG thru eitinject, but I will need a TDT first. I guess it shoud be something like this:
tsp -I ip 239.1.1.1:1234 \
      -P inject -p 20 -b 2000 -tdt.xml \
      -P eitinject -d -f eit.xml \
      -O ip 239.1.1.2:1234
However I have some doubts:
- will TDT tables content (from tdt.xml) be updated according to PCR and bitrate contents in the long run (time/date update) ?
tdt.xml content:
     <TDT UTC_time="YYYY-MM-DD hh:mm:ss"/>
- is there another better way to do it thu the system time ?
Thanks.

lelegard, 2022-05-08:
Although it is well-known that the universal answer is "42", in computing it is in fact "RTFM".

When trying to solve a problem, I would recommend to start by browsing the TSDuck user's guide and especially the list of plugins. The plugin `timeref` is your answer. Read the description of options, you will find everything you need. Note that this plugin only _updates_ existing TDT's. This means that, in the absence of TDT, you need to inject a dummy one using an XML file, as you already do. The same TDT with same content in infinitely cycled, and then updated using the plugin `timeref`.

Litrona, 2022-05-08:
OK, now I got it. 
Thanks for your great help !

closed 2022-05-08 by Litrona