tsduck/tsduck #994: [FEAT] bitrate _monitot with TS and Net bitrate simultaneously.

enricovittorini, 2022-05-06:
#### Is your feature request related to a problem?
Not exactly, but was trying to get every second the gross bitrate of the TS and the net one (without 8191 PID)
The target is to have the two values at exactly the same time and be able to represent them on a chart.

#### Proposed solution:
The output of the bitrate_minitor plugin should report the net bitrate after the whole gross bitrate. Something like
TS Bitrate xxxxz bits/s, Net Bitrate yyyy bits/s.

#### Alternatives:

Using consecitives bitrate_monitor plugin does not guarantee the two values will have the same time.

lelegard, 2022-05-08:
Hi @enricovittorini 

This has been implemented. With periodic messages, the net bitrate is displayed in addition to the global bitrate. With alarm commands, the net bitrate is passed as additional parameter.

As a bonus, it is now possible to monitor the bitrate of a set of PID's (a service for instance). Previously, the bitrate could be monitored either on the full TS or on one PID only.

enricovittorini, 2022-05-11:
@lelegard it is ok. thanks a lot!

closed 2022-09-21 by lelegard