tsduck/tsduck #995: [BUG] PAT doesn't seem to be updated upon restart a stream

lmoody1, 2022-05-06:
#### Bug description:
If I have muxed stream with 2 null input streams, and I use tspcontrol to restart the first one with new settings (new service, and PID), I'm only seeing the output service id as the original.

#### How to reproduce:
I execute the following tsp command (note the first set of merge/pat/pmt plugins for the null input stream are for service id 0xD1 and PID 0xA1, the second set is for service id 0xD2 and PID 0xA2):

    tsp --control-local localhost --control-port 8000 -v -b 41000000 -I null -j -O ip localhost:8999 
    -P merge -j --no-psi-merge --ignore-conflicts -p 0x1-0x20 -d 0x0 -d 0x21-0x1FFE "tsp -v -b 3000000 -I file -i C:\files\file.ts -P regulate" 
    -P merge -j -t "tsp -b 500000 -v -I null -P regulate" 
    -P pat -c -a 0xD1/0x21 
    -P pmt -c -a 0xA1/0x5 -s 0xD1 --pcr-pid 0x1FFF 
    -P merge -j -t "tsp -b 500000 -v -I null -P regulate" 
    -P pat -c -a 0xD2/0x21 
    -P pmt -c -a 0xA2/0x5 -s 0xD2 --pcr-pid 0x1FFF 
    -P regulate

I execute tspcontrol -t localhost:8000 list and this is my result:

    0: -I null
    1: -P merge
    2: -P merge
    3: -P pat
    4: -P pmt
    5: -P merge
    6: -P pat
    7: -P pmt
    8: -P regulate
    9: -O ip

I use the following tspcontrol commands to restart the first set of merge/pat/pmt plugins (lines 2, 3, and 4) with new settings (service id 0xD1 is being changed to 0x8FF4 and the PID from 0xA1 to 0x8E1):

    tspcontrol -t localhost:8000 restart -v 2 -j -t "tsp -b 1000000 -v -I file -i C:\file\input_stream_pid_8E1.ts -P regulate" 
    tspcontrol -t localhost:8000 restart -v 3 -c -a 0x8FF4/0x21 
    tspcontrol -t localhost:8000 restart -v 4 -c -a 0x8E1/0x5 -s 0x8FF4 --pcr-pid 0x1FFF

#### Expected behavior:
The output for the restarted set of plugins should be outputting to service id 0x8FF4

#### Errors and logs:
The output for the restarted set of plugins is still outputting to service id 0xD1

#### Environment:
 - OS: Windows
 - OS version: Windows 10
 - TSDuck full version: 3.30-2710
 - Installation type: official binary TSDuck-Win64-3.30-2710.exe

#### Additional information:
This is the **_really_** weird part, if I then restart the **_second_** set of merge/pat/pmt plugins (lines 6, 7, 8) like this (replacing service id 0xD2 with 0x8FF5, PID 0xA2 with 0x8E2):

    tspcontrol -t localhost:8000 restart -v 5 -j -t "tsp -b 1000000 -v -I file -i C:\file\input_stream_pid_8E2.ts -P regulate" 
    tspcontrol -t localhost:8000 restart -v 6 -c -a 0x8FF5/0x21 
    tspcontrol -t localhost:8000 restart -v 7 -c -a 0x8E2/0x5 -s 0x8FF5 --pcr-pid 0x1FFF

both input streams are outputting to the correct services (0x8FF4 and 0x8FF5)

**_I've attached a Word document describing my steps and screenshots of what I'm seeing (I'm using tsReader to view the output stream)._**

I hope all of this makes sense.

Thanks,

Lee

[Issues restarting null import streams with new parameters.docx](https://github.com/tsduck/tsduck/files/8643258/Issues.restarting.null.import.streams.with.new.parameters.docx)

lelegard, 2022-05-07:
Hi @lmoody1 

There is no bug, this is the expected behaviour:

- In plugin 0 (`null`), there is no PAT.
- In plugin 1 (`merge`), you mention `-d 0x0`. So, you drop the PAT. At this point, there is still nothing in PID 0.
- In plugin 2 (`merge`), the PSI will be merged. Meaning that all occurences of the global tables such as PAT, CAT, SDT, from the main stream will be updated with information coming from the merge stream. Since there is still no PAT in the main stream, it will never be updated.
- In plugin 3 (`pat`), the PAT will be updated, or created (`-c`) if not found in the main stream. Since there is still no PID 0 in the main stream, the PAT will always be created at that point.

Either you let the first merge pass its PAT in transparent mode or you place the `-P pat -c` plugin right after the first merge to enforce the creation of the PAT before the `merge` plugin which is supposed to merge the PSI.

BTW, with a `null` input plugin, you must move the `regulate` plugin in position 1, right after the input plugin. Otherwise, you merge into an unregulated stream.

lmoody1, 2022-05-09:
Thank you for responding.

I've done what you've suggested, move my regulate and added a -P pat -c plugin after the first merge. I also removed the "-d 0x0" option from the first merge (I've tried both with and without this option removed). Here is the new tsp:

    tsp --control-local localhost --control-port 8000 -v -b 41000000 -I null -j -O ip localhost:8999
    -P regulate
    -P merge -j --no-psi-merge --ignore-conflicts -p 0x1-0x20 -d 0x21-0x1FFE "tsp -v -b 3000000 -I file -i C:\files\file.ts -P regulate"
    -P pat -c
    -P merge -j -t "tsp -b 500000 -v -I null -P regulate"
    -P pat -c -a 0xd1/0x21
    -P pmt -c -a 0xa1/0x5 -s 0xd1 --pcr-pid 0x1fff
    -P merge -j -t "tsp -b 500000 -v -I null -P regulate"
    -P pat -c -a 0xd2/0x21
    -P pmt -c -a 0xa2/0x5 -s 0xd2 --pcr-pid 0x1fff

Unfortunately, I'm still getting the same results when I attempt to restart the first set of null merge/pat/pmt plugins (lines 5, 6, 7).

closed 2022-06-02 by lmoody1