tsduck/tsduck #996: [BUG] tspcontrol restart with tables plugin - error

enricovittorini, 2022-05-08:
#### Bug description:
running the following command 👍 
`tspcontrol -t 127.0.0.1:9000 restart  --same  1`

the main tsp process crash with the following error:

```
* Error: tables: line 1: trailing Declaration, invalid XML document, need one single root element
* Error: tables: Main model for TSDuck XML files not found: tsduck.tables.model.xml

seems same bug as #856

#### Environment:
 - OS: Windows 
 - OS version: 10
 - TSDuck 3.31-2737
 - Installation type: official

lelegard, 2022-05-08:
Please provide the complete tsp command. Does the plugin `tables` use JSON output? The bug in #856 related to plugin `psi`.

enricovittorini, 2022-05-08:
yes, use json output:

`tsp --control-port 9000 -I ip 239.0.0.1:5001 -l 20.0.0.100 -P tables --log-json-line --invalid-sections  -O drop`

enricovittorini, 2023-10-30:
@lelegard  any news on this bug? it is still present in verison 3.36-3353. When i run 

`tsp -v -d1 --control-port 3001 -I ip 239.0.0.1:5555 -l 172.16.0.100 -P bitrate_monitor -p 1 -t 1 -P tables --log-json-line --psi-si --pid 18 --pid 20 --invalid-versions  -O drop`

and restart the table plugin
`tspcontrol -t localhost:3001 restart -s 2`

i get:
`Debug: control commands: setting socket receive timeout to 5,000 ms
* control commands: received from 127.0.0.1:55051: restart -s 2
* tables: restarting due to remote tspcontrol
* Debug: tables: loading XML file C:\Program Files (x86)\TSDuck\bin\tsduck.tables.model.xml
* Error: tables: line 1: trailing Declaration, invalid XML document, need one single root element
* Error: tables: Main model for TSDuck XML files not found: tsduck.tables.model.xml
* Debug: tables: debug level set to 1
* Debug: tables: restarted plugin tables, status: false
* Debug: control commands: debug level set to 1
* Debug: control commands: closing socket writer
* Debug: tables: packet processing thread aborted after 0 packets, 125,815 passed, 0 dropped, 0 nullified
* Debug: tables: stopping the plugin
* Debug: drop: stopping the output plugin
* Debug: bitrate_monitor: packet processing thread aborted after 125,822 packets, 125,822 passed, 0 dropped, 0 nullified
* Debug: drop: output thread terminated after 125,815 packets (125,815 output)
* Debug: bitrate_monitor: stopping the plugin
* Debug: ip: stopping the input plugin
* ip: leaving multicast group 239.0.0.1 from local address 172.16.0.100
* Debug: ip: input thread aborted after 125,836 packets
* Debug: control commands: control command thread completed
* Debug: ip: aborting IP input
* Debug: Report logging thread terminated`

thanks!

lelegard, 2023-10-30:
Hi @enricovittorini 

Very sorry for having let this one under the radar for so long. It unfortunately slipped into oblivion...

Too bad, since the fix is a simple one-liner 😞 

Better late than never, it is fixed now.

enricovittorini, 2023-10-31:
Thanks @lelegard !

closed 2023-10-31 by lelegard