tsduck/tsduck #997: [Q] TDT table update with timeref issue

Litrona, 2022-05-08:
I have an SPTS complete file with all SI tables except TDT/TOT. 
I've created a simple tdt.xml file with this content:

<?xml version="1.0" encoding="UTF-8"?>
<tsduck>
  <TDT UTC_time="2020-12-31 00:00:00">
    <metadata PID="20"/>
  </TDT>
</tsduck>

and fired this cmdline:

tsp -I file mpts2.ts 
-P regulate --pcr-synchronous 
-P inject -p 20 -b 2000 tdt.xml 
-P timeref --system-synchronous 
-O .....................

 I got this messages:
 * Warning: timeref: got TDT/TOT PID packet with no complete section inside, cannot update
 
What am I missing ?
Thanks

lelegard, 2022-05-08:
TDT sections shall be alone in their TS packet. It is useless and counterproductive to pack them (unlike EIT, for instance,  which _must_ be packed for obvious bandwidth reasons).

So, you need to add option `--stuffing` in plugin `inject` for the TDT. RTFM for more details.

Litrona, 2022-05-09:
OK thanks done.

closed 2022-05-09 by Litrona