tsduck/tsduck #1000: [FEAT] table plugin to log table if content change but version does not

enricovittorini, 2022-05-10:
#### Is your feature request related to a problem?
Today, only if the table version changes the table is logged. In live it can be useful to monitor such behavior real-time.
When table version does not change but table content does, tables plugin should log the table. 

#### Proposed solution:
tables or psi plugin should keep a signature of the last logged table and keep monitoring. If it changes, log the table

lelegard, 2022-05-10:
Hi @enricovittorini 

As you probably know, this is an invalid behavior. MPEG states that, for long-header sections, the version number shall be changed when the content changes. For a given section number in a given PID, all successive sections with identical table id, table id extension and version must have identical content.

In `tstables` or plugin `tables`, you have various options which may help detecting this, even though this is not exactly what you described. See `--all-sections`, `--no-duplicate` and a few others.

enricovittorini, 2022-05-10:
@lelegard, I doo agree. this is the reason why it is of interest to monitor it. 
It would be great to have such feature one day.

Note: --all-section is not allowed with json output.

lelegard, 2022-06-05:
Hi @enricovittorini 

I have added option `--invalid-versions` in `tstables` and `tables` plugin to track inconsistent section content change without version change. In verbose mode, such events are reported in the final error summary.

The default behaviour, without that option, remains the MPEG-defined one: new tables are triggered only on new version for a given table id and table id extension.

closed 2022-09-21 by lelegard