tsduck/tsduck #1001: [Q] MPTS : How to save multiple PID in different file without calling the stream multiple times

Anas92230, 2022-05-11:
Hello,

I try to find a way to save multiple PID (TXT,data) in different files in MPTS without joinning stream multiple times.

Thanks for help

lelegard, 2022-05-12:
Hi @Anas92230 

Do you mean saving PID 100 in file f100.ts, PID 101 in file f101.ts, etc?

To work on individual PIDs or set of PID's, I recommend a two-step approach: First step, select the packets and assign them a label. Second step, use any plugin on that label only.

The command would be something like that:
~~~
tsp -I ... \
    -P filter --pid 100 --set-label 0 \
    -P filter --pid 101 --set-label 1 \
    -P file --only-label 0 file100.ts \
    -P file --only-label 1 file101.ts \
    -O drop
~~~

Anas92230, 2022-05-12:
This is exactely what i mean.Thank you so much.
it's possible to launch a tsp command on ip stream and call it later for adding filter?
ex:
cmd 1 tsp -I ip XXXX@YYYY:PPP -l IIII  -O drop
and later adding specific filter on cmd1 without launching an another stream?
if it's not possible i'll used your approach.
Thank you

lelegard, 2022-05-12:
You cannot add a new plugin on the fly without stopping `tsp`. The command `tspcontrol` can suspend, resume or restart a plugin, possibly with different parameters, but cannot add a plugin.

Anas92230, 2022-05-12:
ok thank you so much @lelegard 
I close ticket

closed 2022-05-12 by Anas92230