tsduck/tsduck #1003: [Q] Upgrade version 3.25 to 3.30 : tsversion: error while loading shared libraries: libsrt.so.1.4: cannot open shared object file: No such file or directory

Anas92230, 2022-05-12:
Hi,
I use Tsduck version 3.25 with ubuntu 21.10.
when i upgrade version to 3.30 i have this error :
# tsversion --check
tsversion: error while loading shared libraries: libsrt.so.1.4: cannot open shared object file: No such file or directory
when i downgrade to 3.25 all it's ok.
Thank you for helping

moving-digital, 2022-05-13:
hi @Anas92230 

Have you launched the script "install-prerequisites" first ? That's one way to fix your problem.

The other way is to install the libsrt.so.1.4 package manually. Some info [here](https://github.com/tsduck/tsduck/issues/873).

Anas92230, 2022-05-13:
hi @moving-digital 
Thank you for your help i just install "apt install libsrt1.4-openssl" and install 3.30 version and all it's ok now :-)

closed 2022-05-13 by Anas92230

lelegard, 2022-05-13:
Hello @Anas92230 ,

There is somethink weird here. I understand that installing libsrt fixes the issue because this is a prerequisite. But I do not understand how you could have installed tsduck without libsrt being installed first. The installation of tsduck should have failed.

Example using official 3.30 binary:
~~~
$ sudo dpkg -i tsduck_3.30-2710.ubuntu21_amd64.deb 
Selecting previously unselected package tsduck.
(Reading database ... 193542 files and directories currently installed.)
Preparing to unpack tsduck_3.30-2710.ubuntu21_amd64.deb ...
Unpacking tsduck (3.30-2710.ubuntu21) ...
dpkg: dependency problems prevent configuration of tsduck:
 tsduck depends on libsrt | libsrt1 | libsrt1-openssl | libsrt1.4-openssl; however:
  Package libsrt is not installed.
  Package libsrt1 is not installed.
  Package libsrt1-openssl is not installed.
  Package libsrt1.4-openssl is not installed.

dpkg: error processing package tsduck (--install):
 dependency problems - leaving unconfigured
Errors were encountered while processing:
 tsduck
~~~

Of course, you can still use some `--force-all` option in dpkg to force the installation despite the lack of dependency but the error message indicates that it won't work anyway.

@moving-digital, the install-prerequisites script is used to setup the environment to _build_ TSDuck. Once a binary package is produced, the dependencies are enforced by the package management (apt and dpkg on Ubuntu).

Anas92230, 2022-05-13:
hello @lelegard 
previsouly i always have 3.25 version and i used tsversion --upgrade.
After that when i check version with tsversion --version i had this error : 
tsversion: error while loading shared libraries: libsrt.so.1.4: cannot open shared object file: No such file or directory
i used also apt install ./tsduck_3.30-2710.ubuntu21_amd64.deb => same problem.