tsduck/tsduck #1005: [Q] tstables clarification on NIT sections

enricovittorini, 2022-05-16:
tstables is used with std output: the section number is identifies for each section

```
* NIT Actual, TID 64 (0x40), PID 16 (0x0010)
  Version: 27, sections: 6, total size: 5361 bytes
  - Section 0:
  - ...
  - ...
  - ....
```

when json output is used the section information is lost. the section is only present for the transport_stream object through the preferred_section, but not for the network descriptors.

```

        "#name": "transport_stream",
        "#nodes": [
          {
            "#name": "service_list_descriptor",
            "#nodes": [
...
....
        "preferred_section": 0,
..
```.

Am I missing something or is there a problem with the plugin?

command used:
`tstables -p 16 --log-json-line file.ts`

lelegard, 2022-05-16:
This is the intended behaviour, the same for all tables, not only the NIT.

The textual dump and the XML/JSON representations work are different levels of abstraction.

You may see the textual dump as a "disassembler". It dumps the binary structure of the sections. Sections can be dumped individually or grouped to form a complete table, depending on what you dump (`tstables` -> tables, `tstables -a` -> sections, `tstabdump` -> sections).

The XML/JSON structures represent the logical structure of tables, regardless of the way they are implemented as sections. These formats are used for "compiler" / "decompiler" (not "disassembler"). When `tstables` produces an XML/JSON structure, it rebuilds a logical structure from the various sections. You may edit the XML/JSON and recompile it, producing completely new sections.

In the case of the NIT, the `preferred_section` attribute is just a hint that is used by the table compiler as a recommendation to place a TS representation in a given section, if possible, and only if possible. This was required in the past by some national regulators who required that for obscure political reasons ("I want my TS in section #xx").

enricovittorini, 2022-05-16:
@lelegard 
do you think it could be possible to have a json structure that represents the NIT in network_information_sections?
the idea it to have a JSON that can be used to represent exactly what it the table.

lelegard, 2022-05-16:
This is not a matter of NIT, this is a matter of level of abstraction for all tables. The XML/JSON representation targets the table logical level, not the binary implementation level, as segmented in sections. There is no way to do that with TSDuck and no plan to do so.

enricovittorini, 2022-05-16:
that's a pity!  thanks anyway...

enricovittorini, 2022-05-16:
@lelegard , I was thinking twice on this topic.. I think it is important is a environment made of specs to offer a view of the tables as they are generated. I think it would be great to improve this amazing software. hope this is not a huge effort and will be there one day or another... at least before broadcast will disappear from the industry!

lelegard, 2022-05-16:
> offer a view of the tables as they are generated.

That exists, this is the default textual description from `tstables`. You can also save the binary sections in section files or even send each binary section on a UDP port. The feature which is not available is to manipulate individual sections both ways, XML/JSON to binary and binary to XML/JSON. You will realise that this is not possible in the general case. The content of a table is not generated the same way depending on its content, maybe not with the same number of sections. If you could decompile one section as XML/JSON, edit it and recompile it (a typical use case), there is no guarantee that the result will fit in one section. This is why we need to use a global logical view of the table in XML/JSON and then generate the appropriate number of sections. The number and organisation of sections will vary based on the edited modifications.

> at least before broadcast will disappear from the industry!

You can still use TSDuck with multicast IP-TV and HLS streaming.

As far as I am concerned, I have already left the TV industry, broadcast and streaming. Since the beginning of 2022, I work in the microprocessor industry. I now maintain TSDuck only for its user base.

enricovittorini, 2022-05-19:
@lelegard what is the end char of the binary data sent over udp? I mean how do I understand that a the data that makes a table is completed and I can start parsing?

lelegard, 2022-05-20:
> what is the end char of the binary data sent over udp

There is none. For two reasons: 1) binary data, 2) UDP.

There can be no "end char" for binary content since it would be indistinguishable from binary content. And there is no need for "end char" using UDP since UDP is datagram-oriented (vs. stream-oriented with TCP).

So, one UDP message is self-delimited. The message is either a TLV encapsulation (sections, PID and time stamp) or raw sections (see option `--no-encapsulation`). When sections are logged one by one (option `--all-sections`), there is one section in the message. Otherwise, the message contains all sections of the table. Since sections are some kind of TLV structures, each section is clearly delimited. There is no need to have separate processing for sections and tables, the generic processing of tables works in both cases.

~~~
  --no-encapsulation
      With --ip-udp, send the tables as raw binary messages in UDP packets. By
      default, the tables are formatted into TLV messages. 
~~~