tsduck/tsduck #1007: [Q] alpine make installer

enricovittorini, 2022-05-21:
is alpine supported for the "make installer" command?
cose i am getting with 3.31-2742
```
/tsduck # make installer
make: Nothing to be done for 'installer'.
```

lelegard, 2022-05-21:
Hi @enricovittorini 

For Linux systems, only `.rpm` (Red Hat, Fedora) and `.deb` (Ubuntu, Debian) packages can be created.

For other distros, you need to `make -j10 && make install` to install TSDuck directly after building it.

About Alpine, the reason for not having an installer is simple and stupid: I have no idea on how to build a `.apk` package. If you have a pointer to a simple but complete explanation about how to do it, I may consider adding that package too.

enricovittorini, 2022-05-27:
@lelegard 
I spent some time  and i think I found a way to package for alpine by reading:
- https://www.fedux.org/articles/build-alpine-packages-in-a-container.html
- https://wiki.alpinelinux.org/wiki/Creating_an_Alpine_package

below the Dockerfile that will run a docker, build tsduck and create the package:

```
FROM alpine:latest

# install required packages to build "alpine linux" packages
RUN apk add --update --no-cache --no-progress alpine-sdk coreutils bash
RUN apk add --update --no-cache --no-progress sudo g++ linux-headers libsrt

# setup directory for built packages
RUN mkdir -p /var/cache/distfiles
RUN chmod a+w /var/cache/distfiles
RUN chgrp abuild /var/cache/distfiles
RUN chmod g+w /var/cache/distfiles

# setup the abuild configuration
RUN echo 'PACKAGER="Your Name <your@email.address>"' >> /etc/abuild.conf
RUN echo 'MAINTAINER="$PACKAGER"' >> /etc/abuild.conf
RUN echo "%abuild ALL=(ALL) NOPASSWD: ALL" >> /etc/sudoers.d/abuild

# setup the build user in container
RUN adduser -D duckuser
RUN addgroup duckuser abuild

USER duckuser
WORKDIR /home/duckuser/

RUN abuild-keygen -a -i -n
RUN git clone --depth 1 https://gitlab.alpinelinux.org/alpine/aports
WORKDIR /home/duckuser/aports
RUN newapkbuild tsduck
COPY APKBUILD /home/duckuser/aports/tsduck/

WORKDIR /home/duckuser/aports/tsduck
RUN chown duckuser:duckuser /home/duckuser/aports/

#start build process
RUN abuild checksum
RUN abuild -r

#RUN git config --global user.name "duckuser"
#RUN git config --global user.email "duckuser@example.com"
```

this the APKBUILD file: here you need to specify the version to download and build.
**pkgver=3.30
pkgrel=2710**

```
# Contributor: Your Name <your@email.address>
# Maintainer: Your Name <your@email.address>
pkgname="tsduck"
pkgver=3.30
pkgrel=2710
pkgdesc="tsduck analyzer"
url="https://tsduck.io/"
arch="all"
license="BSD 2-Clause"
depends="libsrt"
makedepends="g++ linux-headers"
checkdepends=""
install=""
subpackages="$pkgname-doc"
source="https://github.com/tsduck/tsduck/archive/refs/tags/v$pkgver-$pkgrel.tar.gz"
builddir="$srcdir/$pkgname-$pkgver-$pkgrel"

prepare() {
        default_prepare
}

build() {
        # Replace with proper build command(s)
        cd "$builddir"
        make NOCURL=1 NODEKTEC=1 NOPCSC=1 NOHIDES=1 NOEDITLINE=1 NORIST=1 NOTEST=1
}

check() {
        # Replace with proper check command(s)
        :
}

package() {
        # Replace with proper package command(s)
        install -Dm644 LICENSE.txt "$pkgdir"/usr/share/licenses/$pkgname/LICENSE.txt
        cd "$builddir"
        make DESTDIR="$pkgdir" install SYSROOT="$pkgdir"
}
```

I am using windows and installed DockerDesktop

put the two above file in the same directory, then form a command prompt run
`docker build -t duckgit .`

Once the build is finished,  the apk files are inside the container /home/duckuser/packages/aports/x86_64/

Still not perfect but at least you get an apk file. need to understand how to add properly the tsduck license (for the moment a pkg-doc file is created) and how to build from last commit.

lelegard, 2022-05-30:
Hi @enricovittorini 

Thanks for this contribution. I will try the apk build.

We need to separate the docker and apk subjects. It is true that most containers run Alpine because this is a lightweight distro. But Alpine is a distro in itself and we must be able to build an apk without docker. The idea is to be able to run `make installer` on an Alpine machine and get a TSDuck apk. I have an Alpine VM and will use it for that purpose (I have 10 Linux VM's with 10 different distros...)

There are already different contributed docker builds with TSDuck with direct installation inside the container, not using an apk package.

enricovittorini, 2022-05-31:
@lelegard 
To do it with a VM, we can run the same steps:

Install all dependences:
`apk add --update --no-cache --no-progress  alpine-sdk coreutils bash sudo g++ linux-headers libsrt`

Then:
```
# setup directory for built packages
mkdir -p /var/cache/distfiles
chmod a+w /var/cache/distfiles
chgrp abuild /var/cache/distfiles
chmod g+w /var/cache/distfiles
```
setup the abuild configuration
```
echo 'PACKAGER="Your Name <your@email.address>"' >> /etc/abuild.conf
echo 'MAINTAINER="$PACKAGER"' >> /etc/abuild.conf
```

In /etc/abuild.conf uncomment the PACKAGER and insert your email and uncomment the MAINTAINER  line

```
# PACKAGER and MAINTAINER are used by newapkbuild when creating new aports for
# the APKBUILD's "Contributor:" and "Maintainer:" comments, respectively.
#PACKAGER="Your Name <your@email.address>"
#MAINTAINER="$PACKAGER"
```

`echo "%abuild ALL=(ALL) NOPASSWD: ALL" >> /etc/sudoers.d/abuild`

setup the build user
```
adduser -D duckuser
addgroup duckuser abuild
```

change to duckuser user
```
su duckuser
cd /home/duckuser/
```

```
abuild-keygen -a -i -n
git clone --depth 1 https://gitlab.alpinelinux.org/alpine/aports
cd /home/duckuser/aports
newapkbuild tsduck
cd tsduck/
```

Replace the content of APKBUILD with the file APKBUILD

`chown duckuser:duckuser /home/duckuser/aports/`

start build process
```
abuild checksum
abuild -r
```

Once finished, the package will be in:
`/home/duckuser/packages/duckuser/x86_64`

to install it:
`sudo apk add --allow-untrusted tsduck-3.30.2710-r0.apk`

Some notes on the APKBUILD file:
I createda new variable $duckver as the naming convention of tsduck seems not valid for abuild (3.30-2710).
Maybe someone can help to improve the below file.

in the build() section you can specify the option to bulld tsduck

```
# Contributor: Your Name <your@email.address>
# Maintainer: Your Name <your@email.address>
pkgname="tsduck"
duckver=3.30-2710
pkgver=3.30.2710
pkgrel=0
pkgdesc="tsduck analyzer"
url="https://tsduck.io/"
arch="all"
license="BSD 2-Clause"
depends="libsrt"
makedepends="g++ linux-headers"
checkdepends=""
install=""
subpackages="$pkgname-doc"
source="https://github.com/tsduck/tsduck/archive/refs/tags/v$duckver.tar.gz"
builddir="$srcdir/$pkgname-$duckver"
options="!check toolchain"
prepare() {
        default_prepare
}

build() {
        # Replace with proper build command(s)
        cd "$builddir"
        make NOCURL=1 NODEKTEC=1 NOPCSC=1 NOHIDES=1 NOEDITLINE=1 NORIST=1 NOTEST=1
}

check() {
        # Replace with proper check command(s)
        :
}

package() {
        # Replace with proper package command(s)
        install -Dm644 LICENSE.txt "$pkgdir"/usr/share/licenses/$pkgname/LICENSE.txt
        cd "$builddir"
        make DESTDIR="$pkgdir" install SYSROOT="$pkgdir"
}
```

lelegard, 2022-06-01:
Thanks @enricovittorini for the examples and pointers. I am trying to turn this into something practical in a git work area. However, at first sight, abuild looks very hostile to developers. It seems to be designed for external packagers on dedicated systems.

I also read various articles and Alpine Wiki entries. It seems that the build process involves system-wide configuration (even specific user creation in your example). Also, it seems that the source must always be the URL of a remote archive (the latest release from Github in your example). This is fine for external packagers, this is very inconvenient for developers (ie. me).

What I expect from a packaging system is to build a package from my user account, without specific privilege (even though I can sudo in all my VM's) and using the source files from my local tree (not a release, not even pushed on Github). This is possible with .rpm and .deb packages. If this is not possible for .apk packages, I will leave that to external packagers.

closed 2023-04-22 by lelegard