tsduck/tsduck #1009: [Q] Is it possible to have multiple UDP destination in one MPE PID?

afriza, 2022-05-24:
We have a use case where we need to have a lot of UDP destinations delivered over satellite. If we use 1 PID for each UDP destination,

1. many DVB tuners only support a small number of PID filter
2. some DVB tuners allow passing full MPEG-TS stream, but we'll have big unrelated bandwidth mixed in

Does MPE spec support multiple UDP destinations in a single PID? Or do I need to wrap packets for multiple UDP destinations into a "single" UDP destination and unwrap them on the receiver side?

lelegard, 2022-05-25:
Hi @afriza 

> Does MPE spec support multiple UDP destinations in a single PID?

Yes definitely.

You can find one example in the [TSDuck sample streams repository](https://tsduck.io/streams/?name=africa-dstv-mpe-dvbh). You can view multiple UDP destinations in PID 300 using the following command:
~~~
 tsp -I http https://tsduck.io/streams/africa-dstv-mpe-dvbh/68E_12610.912_H_6710.ts -P mpe --pid 300 --log -O drop 2>&1 | grep dest:
~~~

afriza, 2022-05-26:
Hi @lelegard , thank you for the reply.

Do you mind posting sample command for injecting multiple UDP into 1 MPE PID and the corresponding command for extracting them?

lelegard, 2022-05-30:
Hi @afriza 

You need some professional equipment to inject multiple UDP destination addresses inside the same MPE PID. Using TSDuck, there is no way to inject multiple UDP destination addresses inside the same MPE PID. The plugin `mpeinject` listens to one socket only and forwards the UDP packets in the MPE stream.

To extract one specific UDP destination from an MPE PID with multiple destinations, use the plugin `mpe` with option `--destination`. Read the description of that plugin in the [TSDuck user's guide](https://tsduck.io/download/docs/tsduck.pdf) for more details.

afriza, 2022-09-04:
@lelegard Do you know any manufacturer/brand/distributor/seller of such equipment?

lelegard, 2022-09-08:
Hi @afriza 

I have added the capability to multiplex several input UDP streams into one MPE PID. See the user's guide for details.

afriza, 2022-09-08:
For cross-reference:
- https://github.com/tsduck/tsduck/issues/924
- 15e30151844f6caace1aa053b8282fdca7797f72

picatchoof, 2022-09-14:
> @lelegard Do you know any manufacturer/brand/distributor/seller of such equipment?

Hi Arif .
If you did not succeed to build your own mpe Encapsulator/Deencapsulator using the powerful TSDuck tool, you can buy from IDC (Digisat) the equipment IDC IPE-4000A IP Encapsulator

Or the IP Encapsulator (IPE) MD5600 from Maindata.

But it's somehow expensive because I'm some cases you will be obliged to figure out how to deal with in receiving side. And boxes are also not easy night cheap to find.

Good luck.

afriza, 2022-10-13:
@lelegard We have been testing this feature for more than a week now and have no issues so far. The UDP data also seem to be more stable on receiver side.

Thanks again for the great work!

closed 2022-10-13 by lelegard

maceyldn, 2023-07-04:
Apologies for reopening, looking to get some clarification on usage.

-P mpeinject -b 10000000 --max-queue 10000 -p 7030 -l 192.168.145.145 239.100.200.1:5001 239.199.200.2:5001 239.199.200.3:5001

I'm trying to pull 3 multicasts into the pid 7030. They are approx 2Mbit each.. I see 239.100.200.1 but not the others and they are definatly present on the interface (tcpdump proves it).

Am i missing something?

Thanks
Andy

lelegard, 2023-07-05:
=> duplicate of #1251