tsduck/tsduck #1010: [Q] How to suppress cookies-files when using python3-bindings and HTTP connections

dutchboontje, 2022-05-26:
Hi, 
I am working on an application /w the tsduck python-bindings to transmuxing http-streams into multicast.

There are some 'cookies'-files created in the /tmp directory.

- Is there a way to get ride of these files?

On last check, after hours of running, there were thousends of files named like:  /tmp/tstmp-< hexcode >.cookies
hexcode:  some random hexcode, aka '000B553DAD550000'

Content of those files, are all the same, probably the will be some cookie data when available:
```
# Netscape HTTP Cookie File                                                                              
# https://curl.se/docs/http-cookies.html                                                                 
# This file was generated by libcurl! Edit at your own risk.                    

``` 

- Any suggestion to point me into a direction ?

lelegard, 2022-05-29:
Hi @dutchboontje 

Those files are normally created by the `hls` plugin, not the `http` plugin. Could you confirm that this is the case?

Having cookie files is required in HLS because the HTTP request on the playlist sometimes returns cookies which are required to access the segments. This is not the case with the HTTP plugin.

I just made a test on macOS with a `tsp -I hls` command. I can see the cookies files but it is automatically deleted when the plugin is closed. However, in some cases, I have noticed that the cookies file is not deleted, _even though the debug message "deleting cookies file /tmp/tstmp-00221ABAADF70000.cookies" is displayed_. Looking in the source code, this debug message is immediately followed by the deletion of file. In case of deletion error, an error message is displayed. Since I see no such error message, having the cookie file not deleted is really mysterious. I have to investigate.

In the meantime, could you indicate:
- The operating system name and version
- The TSDuck version
- The exact chain of plugins
- How is the command terminated (end of input, some condition such as plugin `until` or interruption).
- If you use the Python bindings, do you call the method `waitForTermination` of the `TSProcessor` object? This ensures that all plugins are correctly terminated, including the cleanup of the cookies file in the case of the `hls` plugin (even though I have noticed cases where the cookies files is mysteriously not deleted as described above).

dutchboontje, 2022-05-31:
Hi Ielegard,  some answers...
I will have to check /w you patch, still. I hope this week.

Problem found on version 3.31.2722:

```
edwin@streaming-server-tvm01:~$ tsversion --version=all
tsversion: TSDuck - The MPEG Transport Stream Toolkit - version 3.31-2722
Built May 17 2022 - 06:34:31
Using GCC 11.2.0, C++ std 2011.03
System: Ubuntu (Ubuntu 22.04 LTS), on Intel x86-64, 64-bit, little-endian, page size: 4096 bytes
Bitrate: 64-bit floating-point value
Dektec: This version of TSDuck was compiled without Dektec support
Web library: libcurl: 7.81.0, ssl: OpenSSL/3.0.2, libz: 1.2.11
SRT library: libsrt version 1.4.4
RIST library: This version of TSDuck was compiled without RIST support
```

System/OS info:

```
edwin@streaming-server-tvm01:~$ lsb_release -a
No LSB modules are available.
Distributor ID:	Ubuntu
Description:	Ubuntu 22.04 LTS
Release:	22.04
Codename:	jammy
```

Related libCurl version:

```
edwin@streaming-server-tvm01:~$ apt show libcurl4
Package: libcurl4
Version: 7.81.0-1ubuntu1.2
Priority: optional
Section: libs
Source: curl
Origin: Ubuntu
```

Issue/files concerned:

```
edwin@streaming-server-tvm01:~$ uptime; echo; echo -n "#of tstmp-* files..."; ll /tmp/tstmp-*| wc -l
 11:37:59 up 4 days, 17:09,  1 user,  load average: 0.22, 0.21, 0.23
# of tstmp-* files...4101
```

Implementation TSduck /w Python binding, config:

```
# Channel channel1 config
tsp_options = {
	'app_name': 'channel1', 
	'bitrate_adjust_interval': 1000, 
	'buffer_size': 64
	}
input = ['http', '--receive-timeout', '5000', '--user-agent', 'tsduck/3.x', 'http://<masked host>:80/<masked path>']
plugins = [
	['count', '--pid', '0-31', '--negate', '--brief', '--interval', '16620'], 
	['continuity'], 
	['pcrbitrate'], 
	['regulate', '--pcr-synchronous'], ['sdt', '--increment-version', '--provider', 'live', '--service-id', '1', '--name', 'NPO 1']
	]
output = ['ip', '--packet-burst', '7', '--ttl', '5', '--force-local-multicast-outgoing', '--local-port', '1234', '239.1.1.1:1234']
```
Count-plugin is added to monitor traffic of the source.

Implementation TSduck /w Python binding, snip of code:

```
# Create a TS processor using the report.
tsp = tsduck.TSProcessor(rep)

# Set some global TS processing options.
if hasattr(config, 'tsp_options'):
    # just for safety
    if 'app_name' not in config.tsp_options:
        tsp.app_name = "send-origin1-service1"

    for key in config.tsp_options:
        print( f"setting option {key} = {config.tsp_options[key]}")
        tsp.key = config.tsp_options[key] 

tsp.input = config.input
tsp.plugins = config.plugins
tsp.output = config.output

# Run the TS processing and wait until completion.
tsp.start()
tsp.waitForTermination()

# At this stage, there should be some thing going wrong, abondon to restart service
tsp.delete()
rep.terminate()
rep.delete()
```

It's running as an embedded application. A few streams are fired off all /w differend sources and destinations.
No user interaction will be done.  Any actions to be expected are source dis-continuities, although these are supposed to be live-streams.  :{

As you can see it configures on http-inputs module, how come to hls-input ?

Let me know if you needed additional info.

lelegard, 2022-05-31:
Hi @dutchboontje 

You are right, there was another (and deeply hidden) issue with the `http` plugin. In fact it was in the handling of common command line options for web requests. I fixed it too.

To be more specific, the spurious cookies file remained in the `hls` plugin when it was interrupted and in the `http` plugin in all cases.