tsduck/tsduck #1012: dvb-t with dektec

jwsmite, 2022-06-01:
Hi, I have a script that runs ok for simple modulation but it seems unstable after a while audio distortion frames flicking back and forth a bit. Is there a reference for udp -> dektec (DVB-T) and ideally file (looping) -> dektec DVB-T? I'll share my current script, even my AIT seems to not be getting read and run now, I think I'm going round in circles a bit. Tips would be appreciated.

```python
import sys
import subprocess

"""

use live input from udp
add APD
mudulate to dvb-t

note if run with nohup python3 ... & then tsduck subprocesses need to be killed as well as python script

"""

MEDIA_INTERFACE_IP = "10.210.165.170"
AIT_PATCH_FILE = "/home/ubuntu/ait_patch.xml"
PMT_AIT_PATCH_FILE = "/home/ubuntu/ait_pmt_patch.xml"

# first CLI argument is input file transport stream
input_stream = sys.argv[1]

# set modulation parameters in dict in order to reference once for bitrate calc and use again for modulator setup
mod_values = { "bandwidth":             "8-mhz",
               "convolutional_rate":    "7/8",
               "frequency":             "578000000",
               "guard_interval":        "1/32",
               "dmb_constellation":     "64-QAM",
               "modulation":            "DVB-T"}
 
# calculate modulated bitrate and add to dict
tsterinfo_rate = subprocess.run(['tsterinfo',
    "-h", mod_values["convolutional_rate"],
    "-g", mod_values["guard_interval"],
    "-c", mod_values["dmb_constellation"],
    "-s"], stdout = subprocess.PIPE, universal_newlines = True)
 
mod_values["dvb_bitrate"] = tsterinfo_rate.stdout.strip()
 
# without -t option we don't have a PAT to even merge our stream with
# packet burst seems to make big difference to how smooth final playback is
run_tsp = subprocess.run(f'tsp \
                --verbose \
                -d \
                -b {mod_values["dvb_bitrate"]} \
                -I null \
                -P regulate \
                -P merge \
                -t \
                    "tsp -I ip -l {MEDIA_INTERFACE_IP} {input_stream}" \
                -P tsrename --ts-id 0x0001 \
                -P inject --pid 0x0333 {AIT_PATCH_FILE} \
                -P pmt --patch-xml {PMT_AIT_PATCH_FILE} \
                -O dektec \
                -f {mod_values["frequency"]} \
                -m {mod_values["modulation"]} \
                -g {mod_values["guard_interval"]} \
                -r {mod_values["convolutional_rate"]} \
                --dmb-constellation {mod_values["dmb_constellation"]} \
                -s', shell = True)
```

lelegard, 2022-06-01:
Hi @jwsmite,

The bitrate which is expected by the modulator may need to be more precise than what the `regulate` plugin can provide. I would recommend to regulate at a slightly lower bitrate (let's say 2 or 3%) and then use option `--stuffing` in the `dektec` output plugin. Thus, the input of the modulator is slightly less than its output speed and the modulator compensates with null packets.

jwsmite, 2022-06-07:
Hi, Thanks for the tip, I tried calculating a rate 2 (and 3) percent below the tsterinfo calculated value. The --stuffing control may not be working though it seems, after some minutes I start getting dektec: Got DMA underflow. I'm using DTU-315 which I notice isn't mentioned in the manual, I had the initial instability issues with DTU-215 as well

jwsmite, 2022-08-03:
@lelegard I was reading some other issues where you got into this issue of muxing (https://github.com/tsduck/tsduck/issues/394) and I can see that ultimately tsduck is not quite setup for cleanly muxing out like this. It reads as though an alternate approach though is to use a captured ts made with specifc modulation parameters. What would be a minumum duration of capture to use for this and just loop as a ts source to just nullify and overwrite? Would this still ultimately drift out due to the fractional nature of mux rate?