tsduck/tsduck #1016: [Q] discontinuity passing through ip output

barhom, 2022-06-09:
Both tsps are running on the same server.

tsp1)
`
tsp -v -I srt --listener remote-ip:10021 -P continuity -O ip -l 127.0.0.1 239.100.0.17:1234
`

tsp2)
`
tsp -v -I ip -l 127.0.0.1 239.100.0.17:1234 -P continuity -O drop
`

What you can see in this is the first tsp has input srt and output ip/udp/multicast
The second tsp reads from ip/udp/multicast

The weird thing for me is that I am seeing some missing packets only on tsp2.
`
* continuity: packet index: 44,079, PID: 0x0200, missing 5 packets
* continuity: packet index: 44,088, PID: 0x0200, missing 6 packets
* continuity: packet index: 44,093, PID: 0x0200, missing 5 packets 
`

What can the issue be? Only thing that I can think of is if the kernel is dropping packets going to the lo interface - which seems weird.

lelegard, 2022-06-10:
Hi @barhom 

Interesting. I agree that dropping UDP packets on the lo interface is unlikely, although still possible under heavy load.

To investigate the issue, I would recommend to save the TS content before and after the lo interface like this:
~~~
tsp -v -I ip -l 127.0.0.1 239.100.0.17:1234 -P continuity -O file f2.ts
~~~
and
~~~
tsp -v -I srt --listener remote-ip:10021 -P continuity -P file f1.ts-O ip -l 127.0.0.1 239.100.0.17:1234 --enforce-burst
~~~
Start the receiver first to make sure it will capture the TS from the beginning. Use `--enforce-burst` to make sure that all UDP datagrams contain exactly 7 packets. This will be easier to characterize the loss of UDP datagram.

When you see unexpected discontinuities, stop the sender first (Ctrl-C) and then the receiver. "Normally", the files f1.ts and f2.ts should be identical. Track differences using:
~~~
tscmp f1.ts f2.ts --continue --subset --verbose
~~~

If the differences are only missing packets in f2.ts, in chunks of 7 TS packets (or a multiple of 7), you can be sure that this is a loss of UDP datagrams. In that case, this becomes a system & network tuning issue/

barhom, 2022-06-10:
Trying to do what you suggested generated the following,

In console output of receiver
```
* continuity: packet index: 776,713, PID: 0x0200, missing 6 packets
* continuity: packet index: 776,725, PID: 0x1010, missing 1 packets
* continuity: packet index: 776,727, PID: 0x0200, missing 15 packets
* continuity: packet index: 776,729, PID: 0x1010, missing 14 packets
* continuity: packet index: 776,735, PID: 0x0200, missing 11 packets
* continuity: packet index: 776,741, PID: 0x1010, missing 1 packets
* continuity: packet index: 789,103, PID: 0x0200, missing 6 packets
* continuity: packet index: 789,110, PID: 0x0200, missing 4 packets
* continuity: packet index: 789,118, PID: 0x0200, missing 6 packets
* continuity: packet index: 1,396,585, PID: 0x0200, missing 7 packets
* continuity: packet index: 1,396,591, PID: 0x0200, missing 4 packets
* continuity: packet index: 1,396,598, PID: 0x0200, missing 5 packets
* continuity: packet index: 1,440,488, PID: 0x0200, missing 6 packets
* continuity: packet index: 1,440,495, PID: 0x0200, missing 5 packets
* continuity: packet index: 1,440,502, PID: 0x0200, missing 5 packets
* continuity: packet index: 1,453,620, PID: 0x0200, missing 6 packets
* continuity: packet index: 1,453,628, PID: 0x0200, missing 5 packets
* continuity: packet index: 1,453,634, PID: 0x0200, missing 5 packets
* continuity: packet index: 1,594,404, PID: 0x0200, missing 5 packets
* continuity: packet index: 1,594,413, PID: 0x0200, missing 6 packets
* continuity: packet index: 1,594,418, PID: 0x0200, missing 5 packets
* continuity: packet index: 2,162,763, PID: 0x0200, missing 6 packets
* continuity: packet index: 2,162,770, PID: 0x0200, missing 5 packets
* continuity: packet index: 2,162,776, PID: 0x0200, missing 5 packets
*
```

tscmp,

```
tscmp f1.ts f2.ts --continue --subset --verbose
* Comparing f1.ts and f2.ts
* Packet 776,713, missing 7 packets in f2.ts
* Packet 2,511,509: file f1.ts is truncated
* Read 2,511,509 packets, found 1 differences, missing 7 packets in 1 holes
*
```

Now to try to understand this.

How come the console output of receiver showed many different missing packets but tscomp only showed 7packets?
What is happening in the kernel that its dropping packets?

barhom, 2022-06-10:
here is some more debug;

f1.ts
```
tsp -v -I file f1.ts -P continuity -O drop
* file: initial input bitrate is 3,000,000 b/s
* continuity: packet index: 7, PID: 0x0200, missing 5 packets
* continuity: packet index: 14, PID: 0x1010, missing 1 packets
* 
```
f2.ts
```
tsp -v -I file f2.ts -P continuity -O drop
* file: initial input bitrate is 3,000,000 b/s
* continuity: packet index: 7, PID: 0x0200, missing 5 packets
* continuity: packet index: 14, PID: 0x1010, missing 1 packets
* continuity: packet index: 776,713, PID: 0x0200, missing 6 packets
* continuity: packet index: 776,725, PID: 0x1010, missing 1 packets
* continuity: packet index: 776,727, PID: 0x0200, missing 15 packets
* continuity: packet index: 776,729, PID: 0x1010, missing 14 packets
* continuity: packet index: 776,735, PID: 0x0200, missing 11 packets
* continuity: packet index: 776,741, PID: 0x1010, missing 1 packets
* continuity: packet index: 789,103, PID: 0x0200, missing 6 packets
* continuity: packet index: 789,110, PID: 0x0200, missing 4 packets
* continuity: packet index: 789,118, PID: 0x0200, missing 6 packets
* continuity: packet index: 1,396,585, PID: 0x0200, missing 7 packets
* continuity: packet index: 1,396,591, PID: 0x0200, missing 4 packets
* continuity: packet index: 1,396,598, PID: 0x0200, missing 5 packets
* continuity: packet index: 1,440,488, PID: 0x0200, missing 6 packets
* continuity: packet index: 1,440,495, PID: 0x0200, missing 5 packets
* continuity: packet index: 1,440,502, PID: 0x0200, missing 5 packets
* continuity: packet index: 1,453,620, PID: 0x0200, missing 6 packets
* continuity: packet index: 1,453,628, PID: 0x0200, missing 5 packets
* continuity: packet index: 1,453,634, PID: 0x0200, missing 5 packets
* continuity: packet index: 1,594,404, PID: 0x0200, missing 5 packets
* continuity: packet index: 1,594,413, PID: 0x0200, missing 6 packets
* continuity: packet index: 1,594,418, PID: 0x0200, missing 5 packets
* continuity: packet index: 2,162,763, PID: 0x0200, missing 6 packets
* continuity: packet index: 2,162,770, PID: 0x0200, missing 5 packets
* continuity: packet index: 2,162,776, PID: 0x0200, missing 5 packets
* 
```

lelegard, 2022-06-10:
There are two things here.

First, if tscmp shows a hole of exactly 7 packets, then this is for sure one UDP datagram which was dropped. This is quite possible. The kernel does not drop packets, the sender or receiver miss them, usually based on short buffer size. Either use the option `--buffer-size` in the ip input plugin or tune the Linux kernel parameters for UDP. This is no longer a TSDuck question.

Second, more unexpected, is the inconsistency between the result of tscmp and the two analyses on files f1.ts and f2.ts. Please upload these two files on some file transfer site (wetransfer.com or equivalent) so that I can analyze them.

lelegard, 2022-06-11:
Hi @barhom 

Thanks for the files. I just found the reason: This is a UDP datagram ordering issue in the kernel.

First, I requested the files because the `tscmp` output was inconsistent with the continuity analysis of the two files. Indeed, the two files exhibited a bug in `tscmp`. After fixing this bug, the comparison becomes:

~~~
$ tscmp --verbose --continue --subset f1.ts f2.ts
* Comparing f1.ts and f2.ts
* Packet 776,713, missing 7 packets in f2.ts
* Packet 2,511,509: file f1.ts is truncated
* Packet 776,734, missing 1,734,775 packets in f2.ts
* Read 2,511,509 packets, found 3 differences, missing 1,734,782 packets in 2 holes
~~~

With option `--subset`, we consider that the second file is a strict subset of the first one. This is useful to locate missing packets between the two ends of a communication link.

We see that there are 7 missing packets starting at offset 776,713, as you already found. However, we also see that all packets after offset 776,734 in f1.ts are missing in f2.ts, even though there are many packets in f2.ts.

That was the bug in `tscmp`: in `--subset` mode, the last "hole" was not reported when it extended up to the end of the file.

Here, we should probably consider that f2.ts is not a strict subset of f1.ts. In that case, `tscmp` is not able to resynchronize in f1.ts on the content of the next packet in f2.ts.

So, we have another hypothesis: UDP neither guarantees that all packets are delivered nor the order of packets. In our case, the UDP datagrams are maybe received in the wrong order. That may seem weird when there is only a direct link on the local interface through the kernel. However, let's check.

The packet from f2.ts that `tscmp` cannot resynchronize in f1.ts is at packet offset 776,727. The message says 776,734 but this is an offset in f1.ts and there were 7 missing packets previously in f2.ts, so we need to substract 7.

Let's dump that packet:
~~~ 
$ tsp -I file f2.ts --packet-offset 776,727 -P until --packets 1 | tsdump

* Packet 0
  ---- TS Header ----
  PID: 512 (0x0200), header size: 4, sync: 0x47
  Error: 0, unit start: 0, priority: 0
  Scrambling: 0, continuity counter: 3
  Adaptation field: no (0 bytes), payload: yes (184 bytes)
  ---- Full TS Packet Content ----
  47 02 00 13 15 4B 15 8A CC 7D 5A 5F 86 EB 56 04 12 52 72 A0 24 85 D3 C0
  BF DA 45 98 D9 3F 43 79 69 21 60 64 FD 76 68 E6 ED FC 68 6A FA 52 6D E9
  DC 1F 37 81 0B 65 2E 79 B2 E6 01 CD 68 C7 28 E0 09 5B DD 9D 62 7E 12 BB
  9C DC 1E 73 FD E1 A6 F0 06 13 08 1F 74 F7 B4 0D 7E 05 E9 8B 9F 0D 80 AF
  E9 AA A0 06 61 FF D3 60 66 61 B7 9D 5C 59 CD D7 6D 77 37 1F CC EA 03 3B
  1E 31 4F 21 B8 25 E8 63 7B 78 BE 1D 34 48 B9 98 6E D4 22 16 5C 92 76 71
  5B 34 EC 4E 56 C2 80 94 D9 A8 A1 A1 84 B0 22 AF 7A B3 28 17 9E 66 D3 C2
  76 CD 6A 9E 6B 88 A3 32 A5 09 8E 4D 1E 4F 96 A3 11 99 36 F2
~~~ 

Now, let's try to find a matching packet in f1.ts:
~~~ 
$ tsdump f1.ts | grep -B 8 '47 02 00 13 15 4B 15 8A CC 7D 5A 5F 86 EB 56 04 12 52 72 A0 24 85 D3 C0'

* Packet 776,713
  ---- TS Header ----
  PID: 512 (0x0200), header size: 4, sync: 0x47
  Error: 0, unit start: 0, priority: 0
  Scrambling: 0, continuity counter: 3
  Adaptation field: no (0 bytes), payload: yes (184 bytes)
  ---- Full TS Packet Content ----
  47 02 00 13 15 4B 15 8A CC 7D 5A 5F 86 EB 56 04 12 52 72 A0 24 85 D3 C0
~~~ 

We find the packet in f1.ts at offset 776,713, precisely where `tscmp` reported 7 missing TS packets, ie. one UDP datagram. So, we may conclude that the UDP datagrams where not received in the same order as sent.

To confirm this, we can see that the 7 TS packets at offset 776,713 in file f1.ts are found at offset 776,727 in f2.ts:
~~~ 
$ tsp -I file f1.ts --packet-offset 776,713 -P until --packets 7 -O file f1sub.ts
$ tsp -I file f2.ts --packet-offset 776,727 -P until --packets 7 -O file f2sub.ts
$ tscmp f1sub.ts f2sub.ts
$ 
~~~ 
So, the reported 7 missing packets at offset 776,713 were not missed. They were received 14 TS packets later, 2 UDP datagrams later. This kind of local reordering of the UDP datagrams occurs several times later in the file, exhibiting all the discontinuities you see.

I can guarantee that `tsp` uses a strictly sequential pipeline. TS packets are always processed and written in the same order as they were read.

As a conclusion, you cannot rely on your kernel to deliver UDP datagrams in the same order as they are sent. This is allowed by the UDP protocol, just a little bit surprising from a local interface.

Out of curiosity, which operating system do you use and which version?

barhom, 2022-06-13:
@lelegard Thank you for going through all of this!

> Out of curiosity, which operating system do you use and which version?
Debian 11 running in PXE booted environment (all in ram).
```
# cat /etc/debian_version
11.0
# uname -a
Linux ES10 5.10.0-8-amd64 #1 SMP Debian 5.10.46-4 (2021-08-03) x86_64 GNU/Linux
```
>As a conclusion, you cannot rely on your kernel to deliver UDP datagrams in the same order as they are sent. This is allowed by the UDP protocol, just a little bit surprising from a local interface.

I see what you are saying here, but it sounds very surprising. Maybe you need to specify something to the kernel on your outgoing packets from the IP output plugin? This is only a wild guess.
Actually this particular server has about 1gbit/s of "lo" interface multicast traffic - and I haven't experienced this issue with other streams as far as I remember. I need to check again.

I've tried changing the UDP buffers in sysctl to little avail - I guess like you said, the datagrams arent lost they simply came in the wrong order.

Do you have any suggestion on what I should dive into to see how I can solve this issue?

lelegard, 2022-06-13:
> Do you have any suggestion on what I should dive into to see how I can solve this issue?

Not really. I do not know if using multicast on lo interface is a good idea. In theory, this is completely useless since there is no routing and one single node on this "network". I thought that it was just a test to replicate a specific issue on lo interface for simplicity but when you mention 1 Gb/s of "lo" interface multicast traffic, I see that this is more systematic.

However, the datagram order is an generic UDP issue. It should not make any difference whether it is unicast or multicast. Normally. The implementation of multicast is sometimes surprising. You may check if the same multicast address or the same UDP port is used at the same time on the lo interface. If you routinely use a lot of traffic on lo, this could interfere.

This is just a matter of fact, we see UDP packets being sent in one order on lo and being received from time to time in a different order from the same lo interface. This is no longer a TSDuck issue, this is a Linux UDP multicast issue. If you find an explanation, please let us know here.

barhom, 2022-06-13:
I tried something which is at least interesting,

On the same host run the following,
```
iperf3 -s
```
and
```
iperf3 -c 127.0.0.1 -u -b 100000000 -R -l 1316
```

If we have an issue with the kernel giving Out of Order packets iperf would have logged them as follows,
```
example:
iperf3: OUT OF ORDER - incoming packet = 537 and received packet = 538 AND SP = 4
```

Results sending a 100mbit UNICAST shows no out of order packets and 0 drops.
Unfortunately multicast is not supported in iperf3
```
root@ES10:~# iperf3 -c 127.0.0.1 -u -b 100000000 -R -l 1316
Connecting to host 127.0.0.1, port 5201
Reverse mode, remote host 127.0.0.1 is sending
[  5] local 127.0.0.1 port 55050 connected to 127.0.0.1 port 5201
[ ID] Interval           Transfer     Bitrate         Jitter    Lost/Total Datagrams
[  5]   0.00-1.00   sec  11.9 MBytes  99.9 Mbits/sec  0.003 ms  0/9490 (0%)
[  5]   1.00-2.00   sec  11.9 MBytes   100 Mbits/sec  0.009 ms  0/9499 (0%)
[  5]   2.00-3.00   sec  11.9 MBytes   100 Mbits/sec  0.003 ms  0/9500 (0%)
[  5]   3.00-4.00   sec  11.9 MBytes   100 Mbits/sec  0.003 ms  0/9497 (0%)
[  5]   4.00-5.00   sec  11.9 MBytes   100 Mbits/sec  0.007 ms  0/9499 (0%)
[  5]   5.00-6.00   sec  11.9 MBytes   100 Mbits/sec  0.004 ms  0/9498 (0%)
[  5]   6.00-7.00   sec  11.9 MBytes   100 Mbits/sec  0.005 ms  0/9498 (0%)
[  5]   7.00-8.00   sec  11.9 MBytes   100 Mbits/sec  0.008 ms  0/9499 (0%)
[  5]   8.00-9.00   sec  11.9 MBytes   100 Mbits/sec  0.005 ms  0/9498 (0%)
[  5]   9.00-10.00  sec  11.9 MBytes   100 Mbits/sec  0.005 ms  0/9500 (0%)
- - - - - - - - - - - - - - - - - - - - - - - - -
[ ID] Interval           Transfer     Bitrate         Jitter    Lost/Total Datagrams
[  5]   0.00-10.00  sec   119 MBytes   100 Mbits/sec  0.000 ms  0/94978 (0%)  sender
[  5]   0.00-10.00  sec   119 MBytes   100 Mbits/sec  0.005 ms  0/94978 (0%)  receiver
```

barhom, 2022-06-14:
@lelegard since you debugged that there was no packet loss, but only packet-reordering issue, would sending it via RTP on localhost help?
Does tsduck have a buffer where it can do reordering when reading from udp?

barhom, 2022-06-14:
> I suggest to try to use unicast on lo (ie. `-O ip localhost:NNNN` and `-I ip NNNN`). The processing of multicast has always exhibited strange behaviours on different operating systems. It may not change anything but it's worth a try.

I will try this - however this limits me from reading the unicast only by one "reader" afaik which is not something I would like to limit myself to.

lelegard, 2022-06-15:
> I will try this - however this limits me from reading the unicast only by one "reader" afaik which is not something I would like to limit myself to.

I was never convinced of the reliability of multiple readers _on the same machine_ for the same multicast address/port. In principle, it should work. But the beauty of UDP is that nothing is guaranteed, even worse with multicast and the reality is often disappointing. Specifically when working with Linux, macOS and Windows at the same time as I do with TSDuck. There is always one behaving in a strange way.

Again, this is a kernel and network issue, not a TSDuck issue, and I recommend to get help from more appropriate forums.

cbernander, 2022-06-15:
> I will try this - however this limits me from reading the unicast only by one "reader" afaik which is not something I would like to limit myself to.

Actually, you can use multiple readers without multicast by utilizing a localhost broadcast, since the tsduck ip plugin sets the `SO_REUSEPORT` option. I was not able to use `127.255.255.255` as the target IP directly with tsp (`* Error: ip: error sending UDP message: Permission denied`) but by piping the tsp output to socat this setup works fine for me:

```
$ tsp -I file input.ts -P regulate | socat - udp-datagram:127.255.255.255:10000,broadcast
$ tsp -I ip 10000 -P continuity -P bitrate_monitor -p 5 -O file output1.ts
$ tsp -I ip 10000 -P continuity -P bitrate_monitor -p 5 -O file output2.ts
```

barhom, 2023-05-10:
I've found,

* https://unix.stackexchange.com/questions/569682/retransmissions-out-of-order-packets-with-large-1-mb-writes-reads-w-tcp-soc
* https://lore.kernel.org/netdev/7e15f5c4-a9ea-ce26-12e8-1cd1d356fabd@gmail.com/T/#m2511ceb523b5ca88f49b2e1c31d10005df7f18c5

That talks about my issues and it seems the way forward is to enable RPS

echo ffffffff,ffffffff,ffffffff,ffffffff > /sys/class/net/lo/queues/rx-0/rps_cpus

I'll give it a try and let you know if that solves everything