tsduck/tsduck #1018: [Q] How can add maximum_bitrate_descriptor into transport stream on the fly with value from the stream

maxmashnitsky, 2022-06-14:
Hello,

I need to add `maximum_bitrate_descriptor` for the video elementary transport stream in PMT table on the fly with the value from the stream.

Could you please recommend what plugin is better to use and its options?

Thank you in advance.

lelegard, 2022-06-14:
> I need to add maximum_bitrate_descriptor for the video elementary transport stream in PMT table on the fly with the value from the stream.

What do you mean by "with the value from the stream"?

The plugin `pmt` updates a PMT on the fly. Please the TSDuck user's guide for the various options. The specialised options address the usual cases. For any arbitrary modification on the fly, use option `--patch-xml`. You pass the name or content of an XML file which is used as a template for update. This option can be found in several plugins. See section 2.6.4 for details on XML patch files.

maxmashnitsky, 2022-06-15:
@lelegard Thank you for reply.

"with the value from the stream" means analyze the stream and update accordantly the value of the descriptor maximum bitrate.

Yes. I use the `pmt` plugin can apply patch XML. The question is about value for the parameter from the stream.

lelegard, 2022-06-15:
The `analyze` plugin returns an analysis of the TS and all PID's but the bitrate values are average, not maximum.

We may find a way to do that with scripting but it won't be very clean, accurate or robust. This can of operation deserves a new plugin. I recommend to develop a dedicated one. In principle, this is not very complicated:

- Monitor PAT and all PMT's, including version changes.
- Identify all video PID's from the PMT's
- Monitor the bitrate, including maximum bitrate, of each video PID
- Insert/replace the maximum_bitrate_descriptor in the corresponding PMT.

maxmashnitsky, 2022-06-15:
@lelegard Thank you for recommendation