tsduck/tsduck #1021: Output 204 bytes IP stream instead of 188 [Q]

AmaralVini, 2022-06-20:
Is there a way to output a 204 bytes IP stream instead of 188 bytes via command line?
I tried a lot of options, including:

`tsresync --verbose --keep myfile204.ts |  tsp -O ip  --local-address IP:PORT
`

The input file format is RS204, but it seems the command "tsp -O ip" always changes the packets to 188 bytes.

lelegard, 2022-06-21:
Hi @AmaralVini 

A transport stream is a suite of 188-byte packets, nothing else. The input plugin is in charge of extracting the 188-byte packets from the input data, depending on the transport encapsulation. The 16-byte Reed-Solomon trailer is only used by modulators to add error correction. It is not part of the TS. Since some capture devices from demodulated broadcast record the RS trailer, the `file` input plugin can automatically detect it and extract the 188-byte packets. See the documentation of the `file` input plugin in the TSDuck user's guide, especially the `--format` option.

All along the chain of plugins, only 188-byte packets are processed.

The output plugin is responsible for inserting the 188-byte packets in the output format. There is no such thing as RS204 format packets in UDP datagrams. The content of a UDP datagram is always 188-byte packets. Specific protocols such as RTP add a header at the beginning of the UDP payload but the rest is made of 188-byte packets.

If your UDP stream is read by some equipment which injects it into a modulator, this is the responsibility of the modulator to automatically add the Reed-Solomon trailer.

AmaralVini, 2022-06-21:
Hello, I appreciate your reply. 
However, when you say "there is no such thing as RS204 format packets in UDP datagrams", I'd like to convince you that there are RS204 format packtes in UDP datagrams. 
I have several TV multiplexers that we can choose the UDP or RTP output, 188 or 204 bytes. I copied the screen below.
[image]

We can send this RS204 stream output, via UDP, to the modulator, but we also can receive the 204 bytes UDP stream using VLC, tsduck, Dektec TS Analyzer,  or any IRD with IP input. I attached the Dektec screen showing the UDP IP stream received and the packet size of 204 bytes.

[image]

This 204 TS over IP format is used by brazilian and japanese standards of TV Digital, ISDB-T / Tb.  The modulators need to receive the BTS - Broadcast Transport Stream - ready to use, that means a 204 bytes TS with a specific bitrate of 32,507,936 bps.  Some modulators can receive an process packets of 188 bytes, but others can't.
 
As tsduck is ready for these standards in some ways, I tought it can handle the 204 bytes packet size for IP output. And also a given bitrate. 

Is it possible using the tsduck C++ library? 

This IP output format could be an enhancement for tsduck?

lelegard, 2022-06-22:
OK, but this still does not make sense, technically speaking.

The 16-byte Reed Solomon trailer is the outer FEC layer of a broadcast modulation system such as DVB-T/S/C, ISDB, ATSC and the like . What is the technical purpose of using it on non-modulated transport? When you send a TS to a modulator, this is the role of the modulator to add the 16-byte Reed Solomon trailer. You do not even need to provide one and if you provide one it will be ignored and replaced.

To be clear, a transmission system is made of layers and encapsulation of layers. This structure must be followed.

- Broadcast system layers: carrier, modulation, TS, content (PES and signalization).
- IP TV systems: IP, UDP, [RTP, ] TS, content (PES and signalization).

The 16-byte Reed Solomon trailer is part of the modulation layer (or pre-modulation as outer FEC) of a broadcast system. There is no room for it in the UDP transport. You see that this is used somewhere by some system. This is a fact. But this remains a technical kink.

Using 204-byte packet in UDP is consequently not "standard" at all. At best, it is a weird feature of a product to provide a completely useless 16-byte overhead because someone requested it, probably without understanding the uselessness of it.

> Is it possible using the tsduck C++ library?

Certainly. You can use the UDP classes to send whatever you want. It is up to you to build UDP datagrams containing TS packets with the format you like.

> This IP output format could be an enhancement for tsduck?

Everything is possible and, over the years, TSDuck integrated supports for multiple bugs or insane usages we see in the field.

But what should that RS trailer contain? A full Reed-Solomon error correction scheme? Or just a dummy placeholder?

lelegard, 2022-06-23:
> > This IP output format could be an enhancement for tsduck?
>
> Everything is possible and, over the years, TSDuck integrated supports for multiple bugs or insane usages we see in the field.

Option `--rs204` has been added in the `ip` output plugin. Consider it as part of the set of options to support "bugs and insane usages" since the outer FEC layer of a broadcast modulation should be independent of the UDP transport.
~~~
  --rs204
      Use 204-byte format for TS packets in UDP datagrams. Each TS packet is
      followed by a zeroed placeholder for a 16-byte Reed-Solomon trailer.
~~~

The 16-byte trailer contains only zeroes since computing the actual Reed-Solomon FEC is the responsibility of the modulator.

lelegard, 2022-06-24:
> Option --rs204 has been added in the ip output plugin. 

Version 3.31 is just out. This feature was the last one to make its way in 3.31.

AmaralVini, 2022-06-27:
I agree, adding 16-byte trailer with only zeros makes no sense technically.

But looking deeply, I realized that the 16-byte trailer of the BTS is not the same 16-byte trailer used for the Reed Solomon FEC layer (in the modulation layer). 

For digital terrestrial TV (ISDB-T and Tb), before sending the MPEG2-TS to the modulator, we need to insert up to 8 bytes of ISDB-T information, used by the modulator, and 8 bytes of RS-parity (optional) in each packet. And then, to stuff the stream to achieve a constant bit rate of 2048/63 =  32,507,936.51 bps (based on the IFFT sampling frequency). 

In fact, using optimal parameters (64-QAM, FEC 7/8, 1/32 guard interval), the maximum useful data is about 23 Mbps for a 6MHz channel. So, anyway, there is huge waste of the network bandwith. 

But this is the ISDB standard. And it's not such an insane usage, considering all TV transmitters in Brazil, Japan and other 13 countries receive data in this BTS format, with 204 bytes and ~32,5 Mpbs. 

The most important part of the 8-byte ISDB-T information trailer is the packet hierarchical layer (layer A, B or C) - each layer can use different modulation scheme. And each PID can be in different layer. And also some flags, as the emergency broadcast flag (for emergency alerts) and the flag to mark the packet as IIP, containing the TMCC information - used by modulator but no transmitted. 

We can see the full specification in ARIB STD-B31:
[image]
https://www.arib.or.jp/english/html/overview/doc/6-STD-B31v1_6-E2.pdf

I looked into the binary output of a ISDB-T multiplex device. In fact, out of the 16 bytes, only the first 4 bytes are used with ISDB Information. All the other 12 bytes are 0xFF, as you can see in the image below (each line is a packet with 204 bytes, we just can see the last bytes of each packet).
[image]

> Version 3.31 is just out. This feature was the last one to make its way in 3.31.

Great job, thank you!! I tested, the Dektec TS Analyzer could recognize the 204 packet format!
[image]

However, The TS Analyzer couldn't identify the ISDB information below (IIP packets with TMCC information). 
[image]

I also tested send the UDP 204 bytes to a ISDB-Tb transmitter.  This is the aim. It coudn't work yet, need the IIP and hierarchical layer.

The fixed bit-rate of the BTS also can be a problem. In the trace bar of the TS Analyzer we can see that it's not constant. 
Is it possible to set the output bitrate instead of calculate it from the PCR?

lelegard, 2022-06-28:
The 16-byte trailers in the input file are, in your case, part of the ISDB-T modulation layer and are not part of the transport stream layer. `tsp` is a transport stream processor, it ignores modulation layers. Interface with modulation and modulators are implemented in dedicated output plugins which directly interact with modulator devices such as the `dektec` or `hides` plugins.

> The fixed bit-rate of the BTS also can be a problem. In the trace bar of the TS Analyzer we can see that it's not constant.
Is it possible to set the output bitrate instead of calculate it from the PCR?

It depends on the `tsp` command you use. Use `tsp --bitrate XXXXX` to use a specific bitrate value. However, be aware that this is the playout bitrate, not the processing bitrate. If you want to broadcast a file at playout rate, please read the description of the `tsp` command in the TSDuck user's guide.

Finally, I can only repeat what I already wrote: it is not rational to ask for a modulation layer (whether it is DVB or ISDB) into UDP datagrams. If you are not convinced, just look at one example from STD-B31. How would you build the 8-byte "multiplex position" structure when it contains fields such as _frame_indicator_, defined as "During even frame (W0) of the OFDM frame, it is ‘0.’ During odd frame (W1) of the OFDM frame, it is ‘1.’". This kind of information has sense only in the context of a modulation. And most other fields in this structure have similar adherence to the modulation context.

Then, what should we specify? Something like: please add a partial ISDB-T feature in UDP datagram where you fill a structure with that relevant field and that dummy field. Maybe there is a specific situation in Brazil with a specific requirement for an hybrid UDP/IP + ISDB-T output. In that case, a dedicated plugin would be appropriate.