tsduck/tsduck #1023: How to check scte marker present in TV hls url?

kapilgpt06, 2022-06-21:
I have a tv HLS URL and I want to check scte-35 marker is present or not?
Please share any command or solution
Thanks!

lelegard, 2022-06-21:
Hi @kapilgpt06 

If you mean SCTE-35 markers as splice information tables in the TS, then use:
~~~
tsp -I hls URL -P splicemonitor -O drop
~~~
Please read the TSDuck user's guide for a description of the possible options of each plugin.

If you mean SCTE-35 markers as directives in the HLS playlists, then TSDuck cannot do that. You need an HLS analyzer, not a TS analyzer such as TSDuck.

Note that there are at least three different "incompatible standards" (an oxymoron) for SCTE-35 in HLS playlists. This has already been discussed here in past threads. But these markers are just text lines in playlists which are text files. So, it is relatively easy to grab them. But you need to know which of the 3+ "standards" is used by your HLS provider.

kapilgpt06, 2022-06-22:
HI @lelegard 
Thanks for response
I tried this command with hls URL
`URL : https://x.xx.xx/bpk-tv/xxx/output01/index.m3u8?__hdnea__=st=1655905164~exp=1655991564~acl=/*~hmac=cff6e3bbe27447bda83eb8a3a494236164f998834e36be95e37013b80c37f44c`

but getting a error
`* Error: hls: synchronization lost after 0 packets, got 0xF0 instead of 0x47`

lelegard, 2022-06-23:
Hi @kapilgpt06 

Your stream is not a transport stream. A TS is a suite of 188-byte packets, each packet starting with 0x47. The announced content length in the HTTP headers is 189888 bytes, which is not a multiple of 188. The binary content of the stream is dumped and it is clearly not a TS (it would start with 0x47 and 0x47 would be found every 188 bytes).

This is why the error message is "synchronization lost after 0 packets, got 0x55 instead of 0x47". From the beginning of the data, there is no TS packet.

This is something inconsistent in the HLS service. The playlist announces a segment named "xxxxx-audio_98835_hin=98800-video=176400-275986590.ts", with .ts being the standard extension for TS files. The HTTP header for the media segment announces "Content-Type: video/MP2T", meaning TS. But the content is not a TS.

This is an issue to be discussed with the content provider. 

In general, while HLS originally supported TS format only, a few additional formats have been added later, most notably fMP4. However, in that case, the MIME type for the segment shall reflect the actual stream type.

Please note that TSDuck is a TS manipulation tool, not an HLS analyzer. It uses HLS as one of the many input forms for TS. If an HLS stream provides another media format, TSDuck cannot use it.

themoog, 2022-09-28:
@lelegard suppose I do have a native TS file recorded on a dektek and I wanted to see "SCTE traffic" (for example, keep alives), would I see that using splicemonitor plugin? Im running a . TS through it now and it's returning nothing to the command line.

If I was to see something are there any example outputs that could be shared?

Thanks in advance

Neil

themoog, 2022-09-28:
@lelegard sorry solved it needed the -a flag! Great utility as always!