tsduck/tsduck #1027: Is there a way to fix a corrupted videofile using TSDuck utilities? [Q]

HellbringerOnline, 2022-06-25:
Hello. I have a livestream recording that I received using the streamlink, but due to connection problems some of the videofile packets got corrupted. I can see it with ffmpeg.
When I try to extract a lossless AAC stream, ffmpeg discards corrupted packets and the output audio is shorter than the original.
Can I fill the packets with empty data (silence) to keep the original length of the audio and have silence in places where the packets are broken? What utility can I do this?
How can I get information on which part of the video (timing and duration) the data is corrupred?

lelegard, 2022-06-26:
Hi @HellbringerOnline 

This is not possible using TSDuck. TSDuck works on the transport stream layer, not on the audio/video content (except basic analysis). 

I am not aware of a tool which can do that automatically. In the general case, you need to define "corrupted". Physiologically, this is when the sound you hear is bad. But, is it possible for a tool to detect it at audio syntax level? Using some formats, some kinds of corruptions may result in an invalid stream syntax. But when the audio payloads are just binary, "corrupted" does not mean anything in the absence of error detection codes. Ultimately, PCM, the simplest uncompressed sampled audio, is never "corrupted", this is just raw binary content. To automatically detect sound "corruption", in that case, you would need some AI.

If the corruptions are occasional and on editable media files, you need manual A/V editing using any A/V manipulation tool.

HellbringerOnline, 2022-06-26:
> I am not aware of a tool which can do that automatically. In the general case, you need to define "corrupted".

The package is corrupted - so writes ffmpeg. 
I do not mean that some bytes in the video stream or audio stream are corrupted. With the streamlink, I can download livestream fragments. Each fragment has a duration of 1-2 seconds and I can play them independently. Then these fragments are written to one file with the headers saved. But some of these fragments download with errors or do not download completely. But the streamlink continues to write data. My task is to find damaged or missing fragments and replace them with fragments with empty video and sound.

I am attaching some reports for better understanding:
```ffmpeg -err_detect ignore_err -i q.mp4 -c copy -f null -
Input #0, mpegts, from 'q.mp4':
  Duration: 01:43:51.93, start: 80.032000, bitrate: 6277 kb/s
  Program 1
  Stream #0:0[0x100]: Audio: aac (LC) ([15][0][0][0] / 0x000F), 48000 Hz, stereo, fltp, 130 kb/s
  Stream #0:1[0x101]: Video: h264 (High) ([27][0][0][0] / 0x001B), yuv420p(tv, bt709, progressive), 1920x1080 [SAR 1:1 DAR 16:9], 60 fps, 60 tbr, 90k tbn
  Stream #0:2[0x102]: Data: timed_id3 (ID3  / 0x20334449)
Output #0, null, to 'pipe:':
  Metadata:
    encoder         : Lavf59.20.101
  Stream #0:0: Video: h264 (High) ([27][0][0][0] / 0x001B), yuv420p(tv, bt709, progressive), 1920x1080 [SAR 1:1 DAR 16:9], q=2-31, 60 fps, 60 tbr, 90k tbn
  Stream #0:1: Audio: aac (LC) ([15][0][0][0] / 0x000F), 48000 Hz, stereo, fltp, 130 kb/s
Stream mapping:
  Stream #0:1 -> #0:0 (copy)
  Stream #0:0 -> #0:1 (copy)
Press [q] to stop, [?] for help
[mpegts @ 00000220b8dbcec0] Packet corrupt (stream = 1, dts = 39058470).= 250x
q.mp4: corrupt input packet in stream 1
[mpegts @ 00000220b8dbcec0] Packet corrupt (stream = 1, dts = 174418470).= 258x
q.mp4: corrupt input packet in stream 1
[mpegts @ 00000220b8dbcec0] Packet corrupt (stream = 1, dts = 202858470).= 259x
q.mp4: corrupt input packet in stream 1
[mpegts @ 00000220b8dbcec0] Packet corrupt (stream = 1, dts = 255238470).= 258x
q.mp4: corrupt input packet in stream 1
[mpegts @ 00000220b8dbcec0] Packet corrupt (stream = 1, dts = 277018470).= 258x
q.mp4: corrupt input packet in stream 1
[mpegts @ 00000220b8dbcec0] Packet corrupt (stream = 1, dts = 347938470).= 258x
q.mp4: corrupt input packet in stream 1
[mpegts @ 00000220b8dbcec0] Packet corrupt (stream = 1, dts = 442438470).= 258x
q.mp4: corrupt input packet in stream 1
frame=369058 fps=15059 q=-1.0 Lsize=N/A time=01:42:51.92 bitrate=N/A speed= 252x
video:4507255kB audio:98095kB subtitle:0kB other streams:0kB global headers:0kB muxing overhead: unknown```

===============================================================================
|  TRANSPORT STREAM ANALYSIS REPORT                                           |
|=============================================================================|
|  Transport Stream Id: .......... 1 (0x0001)  |  Services: .............. 1  |
|  Bytes: ..................... 4,889,750,280  |  PID's: Total: .......... 5  |
|  TS packets: ................... 26,009,310  |         Clear: .......... 5  |
|     With invalid sync: .................. 0  |         Scrambled: ...... 0  |
|     With transport error: ............... 0  |         With PCR's: ..... 1  |
|     Suspect and ignored: ................ 0  |         Unreferenced: ... 0  |
|-----------------------------------------------------------------------------|
|  Transport stream bitrate, based on ....... 188 bytes/pkt    204 bytes/pkt  |
|  User-specified: ................................... None             None  |
|  Estimated based on PCR's: ................ 6,295,628 b/s    6,831,426 b/s  |
|  Selected reference bitrate: .............. 6,295,628 b/s    6,831,426 b/s  |
|-----------------------------------------------------------------------------|
|  Broadcast time: ............................... 6213 sec (103 min 33 sec)  |
|-----------------------------------------------------------------------------|
|  Srv Id  Service Name                              Access          Bitrate  |
|  0x0001  (unknown) .................................... C    6,294,883 b/s  |
|                                                                             |
|  Note 1: C=Clear, S=Scrambled                                               |
|  Note 2: Unless specified otherwise, bitrates are based on 188 bytes/pkt    |
===============================================================================

===============================================================================
|  SERVICES ANALYSIS REPORT                                                   |
|=============================================================================|
|  Global PID's                                                               |
|  TS packets: 3,077, PID's: 1 (clear: 1, scrambled: 0)                       |
|-----------------------------------------------------------------------------|
|     PID  Usage                                     Access          Bitrate  |
[... 67 lines cut ...]
|  Single Service PID        Transport:                Discontinuities:       |
|  Bitrate: ....... 745 b/s  Packets: ......... 3,077  Expected: ......... 0  |
|  Access: .......... Clear  Adapt.F.: ............ 0  Unexpect: ......... 8  |
|                            Duplicated: .......... 0  Sections:              |
|                            PCR: ................. 0  Unit start: ... 3,077  |
[... cut ...]
|  Service: 0x0001 (1) (unknown)                                              |
|-----------------------------------------------------------------------------|
|  0x02 (2, PMT), TID ext: 0x0001 (1)                                         |
|      Repetition: .... 2020  ms  Section cnt: .... 3,077                     |
|      Min repet.: ..... 975  ms  Table cnt: ...... 3,077                     |
|      Max repet.: .... 2511  ms  Version: ............ 0                     |
===============================================================================

lars18th, 2022-06-29:
Hi @HellbringerOnline ,

As commented by @lelegard the TSDuck toolkit can't solve your problem. The reason is because even your problem is related to _transport_ errors (missing or corrupted packets), the fix requires to be at the _decoding_ level. Therefore your tool is the ffmpeg.

To fix audio errors with ffmpeg you need to use the parameters `re` and `aresample=async=1`. And of course, you need to reencode the audio (that's decode it and after encode the result, because you need to replace the errors with silence). The former parameter enforces to read at the speed of the incoming transport (but don't use it with realtime sources!). And the later could help fixing the missing/erroneous packets.

I recomend that you read the full ffmpeg documentation. It has a lot of examples.
Regards.

HellbringerOnline, 2022-06-29:
> I recomend that you read the full ffmpeg documentation. It has a lot of examples.

Thanks for the answer, I already found a solution to how to insert silence in place of broken packets without loss using the segment command for ffmpeg. Now I'm trying to automate the process using PowerShell.

closed 2022-06-29 by HellbringerOnline