tsduck/tsduck #1028: [Q] Embed Document (Pdf, Jpg, Docx in DVB) and generate TS file as output

picatchoof, 2022-06-29:
Hello All,
Can you please advise if we can use TSDuck (TSP, MPE, File,...) to embed data (HLS video streams, Pdf, Bin, Docx, Jpg,...) to be sent as DVB protcol.
We manage to generate an MPEG-TS file to be sent over ASI and or IP ports toward modulator in order to be broadcasted as DVB-S2. at the end state this data will be used in Satellite DVB-S2 receivers (Boxes) for upgrade OTA purposes or for file delivery issues.

Best regards.

lelegard, 2022-06-29:
Hi @picatchoof 

As usual, this is a matter of protocol. There is no "file in TS" standard but there are many ways to insert data. 

For OTA upgrades, the DVB method is a DSM-CC carousel with preformatted DSI, DII and DDB tables. There is an example in section 5.2.15 of the TSDuck user's guide.

For generic file transfers, you must first write a technical specification of what you want to achieve in terms of protocols and how they interact. Once you precisely know what you need to achieve, TSDuck may help.

You may define your own protocol, your own tables or use some standard encapsulation. Maybe you want to use the FLUTE protocol or similar to encapsulate files into UDP datagrams and then insert those datagrams into the TS. You would need some dedicated FLUTE tool to format the UDP datagrams then the TSDuck `mpeinject` plugin will do the insertion (see section 5.2.21).

picatchoof, 2022-06-30:
@lelegard 
Dear lelegard Hello.
thnx a lot for your quick reply.
I tried to make a breif design of what we expect from TSDuck as powerfull tool.
So if possible to take a look at this attached PDF and correct us if we did wrong.
BTW if possible to suggest us some file transfer tools that we may test for such purpose it will be really greatfull.
Also if **UDP** protocol will be easier and better, especially within the TSDuck processing plugins, we may use it instead of **HTTP** that we put on the design.
[TSDuck_Casting_Server.pdf](https://github.com/tsduck/tsduck/files/9019850/TSDuck_Casting_Server.pdf)

Best regards.