tsduck/tsduck #1029: playback on dektec modulator results in wrong clock time

st-channel24, 2022-07-01:
Hello,

We are having a large collection of transport stream files that we use for testing new TV protos, using Dektec players. (e.g. DTU-315)
We use tsduck via windows cmd-shell, but that one is generated via a python script (with a GUI) that adds some user friendliness :
- it tries to mimic the functionality of Dektec's StreamXpress, including freely setting modulation standard and modulation parameters (which could differ from the original ones),
- the used settings are stored for later re-use.
Practically it means the user has only to watch that input bitrate <= output bitrate.
Both are displayed in the GUI as :
- input bitrate : from tsbitrate
- output bitrate : from a look-up table (and scaling, i.e. for single carrier modulation to scale unto the desired symbol rate).
What I want to obtain is actually an equivalent of the "RMX" tick-mark in StreamXpress.

Thus far, I used a similar tsp processing sequence for all the standards (single carrier DVB-C/S/S2 and well as OFDM DVB-T/T2).
It is illustrated by the following DVB-C example

tsp -v --add-input-stuffing 1/1 -I file "inputfile.ts" -i 
-P limit --bitrate 25324705 
-P timeref --start system 
-O dektec  --device 0 --frequency 482000000 --level -350 --modulation 16-QAM  --bitrate 25324705

remark that :
- the bitrate for the given input file is 14929414
- the bitrate of 25324705 (in -P limit and in -O dektec) corresponds to DVB-C 16QAM 6.875 MS/s
- the amount of stuffing (e.g. 1/1) is chosen by the python script, such that bitrate after stuffing >= output bitrate
- TDT is changed to system time of the streamer PC
- the input file is rather short (61 s) but it is put in a loop over a (very) long period of time.

While on the TV the first impression is that this is working, some users reported that after time, the time indication on the TV is lagging on the PC-clock 
(e.g. more than 2hrs on a total time of 14hrs. Originally reported on DVB-T, but seen on other standards also, while with StreamXpress the clock remains correct).

For DVB-T there is an easy solution : 
- leave out input stuffing, leave out -P limit and leave out bitrate specification on -O dektec.
For DVB-T2, this does not work as I obtain a signal with much smaller Bandwidth than the expected 8 MHz (from spectrum analyser), which cannot be received by the TV.
For DVB-C/S/S2 I don't have a solution either.
I would like to have -if possible- some understanding why this is going wrong as well as a proposal for a better solution.
It would also be nice to have a quicker diagnosis (e.g. debug mode, or some extra processors) as testing any new configuration is very time consuming due to the
long looping that is required now.
I am aware about "regulate", but I don't know how to use it in combination with input stuffing (and keeping PCR correct) as I think it might be required for DVB-C/S/S2 [](url
[log.txt](https://github.com/tsduck/tsduck/files/9028753/log.txt)
)

ntmedia2, 2023-04-12:
hi [st-channel24] may i try your pyton gui i can help you to testing and improve it

ntmedia2, 2023-04-12:
> Hello,
>
> We are having a large collection of transport stream files that we use for testing new TV protos, using Dektec players. (e.g. DTU-315) We use tsduck via windows cmd-shell, but that one is [...]
 may i try your pyton gui i can help you to testing and improve it