tsduck/tsduck #1030: [Q] tsduck output versus tsreader pro forwarding of tsduck output

lmoody1, 2022-07-06:
I have a situation. I'm trying to do a software download using tsDuck to a set-top box using multicast. When it's running, the download can be seen on our network so I know it's actually being broadcast to the correct ip address.

My problem is, if I output the stream directly from tsDuck to the ip address for a set-top box, the box isn't able to process it. However, if I output the stream to a different address then use TSReader to receive that stream and then forward it to the ip for the set-top box, the box is able to process it.

I know I'm technically asking about how tsReader works, but I was hoping that someone might have had experience with this and can provide me settings for my tsDuck command that can get it to emulate whatever tsReader is doing with the stream.

I've looked through all of the settings in TSReader and I'm not seeing how it is transforming the stream. I've compared the output settings for the tsp command and have attempted to make sure they match the settings of tsReader, but I've not been able to determine how they are different.

**So, this works:**

    tsp --control-source localhost --control-port 8000 -v -b 41209008 -I null -j -O ip 237.XX.XX.XX:XXXX -l 10.XX.XXX.XX -t 5
    -P regulate
    -P merge -j --no-psi-merge --ignore-conflicts -p 0x1-0x18 -d 0x0 -d 0x20-0x1ffe "tsp -v -b 3000000 -I file -i C:\files\file1.ts -P regulate"
    -P inject C:\MuxwebSourceTables\PAT_Inject_File.xml --pid 0 --bitrate 15000
    -P inject C:\MuxwebSourceTables\PMT_Inject_File.xml --pid 0x21 --bitrate 15000
    -P mux -b 2000000 C:\files\file2.ts
  
**Then  from tsReader Pro, intercept 237.XXX.XX.XXXX:XXXX and forward it to 239.XXX.XX.XXXX:XXXX**

**This does not work:**

    tsp --control-source localhost --control-port 8000 -v -b 41209008 -I null -j -O ip 239.XX.XX.XX:XXXX -l 10.XX.XXX.XX -t 5
    -P regulate
    -P merge -j --no-psi-merge --ignore-conflicts -p 0x1-0x18 -d 0x0 -d 0x20-0x1ffe "tsp -v -b 3000000 -I file -i C:\files\file1.ts -P regulate"
    -P inject C:\MuxwebSourceTables\PAT_Inject_File.xml --pid 0 --bitrate 15000
    -P inject C:\MuxwebSourceTables\PMT_Inject_File.xml --pid 0x21 --bitrate 15000
    -P mux -b 2000000 C:\files\file2.ts

I hope this all makes sense.

Thanks

lelegard, 2022-07-07:
Hi @lmoody1 

There are many possibilities. This could be a matter of network topology. You force the multicast output to a specific local interface. TSReader seems reachable from that interface. What about the STB?

Where is TSReader running? On another system or same system as tsp?

You specify TTL = 5 on output. This looks fine on a local test. But could the STB be away by more than 5 hops? Assuming that TSReader may specify a larger TTL.

Maybe the STB is not reacting properly if there are less than 7 TS packets per UDP datagram. Try `-O ip --enforce-burst`.

I would also suggest to use Wireshark and compare the UDP datagrams from TSReader and from tsp. Is there any difference? Try this on the same LAN as the STB. If you do not see the datagrams from tsp, this could be a routing or TTL issue.

lmoody1, 2022-07-26:
Hi @lelegard,

Adding the --enforce-burst switch seems to have resolved our problem.

Thank you for the help.