tsduck/tsduck #1032: [Q] Building with NOPCSC=1 and NOEDITLINE=1, not working as expected?!

bugre, 2022-07-13:
Hi,
I building tsduck on a ubuntu 20.04, and based on what i understood from `https://tsduck.io/doxy/building.html` i'm using make with options `NOPCSC=1` and `NOEDITLINE=1`

```
git clone https://github.com/tsduck/tsduck.git
cd tsduck
bash -x scripts/install-prerequisites.sh
make -j10 NOPCSC=1 NOHIDES=1 NODEKTEC=1 NOEDITLINE=1 NODTAPI=1 NORIST=1
```

I was expecting that i would not need the libpcslite nor the libedit2. But, it seems i missed something or did specify the make options in a wrong format, because checking the generated 'tsp' binary with ldd, shows that it still needs those libraries.

```
 ldd /usr/local/bin/tsp
	linux-vdso.so.1 (0x00007ffc0c4e2000)
	libtsduck.so => /usr/local/lib/libtsduck.so (0x00007fa4987e0000)
	libstdc++.so.6 => /lib/x86_64-linux-gnu/libstdc++.so.6 (0x00007fa4985fb000)
	libm.so.6 => /lib/x86_64-linux-gnu/libm.so.6 (0x00007fa4984ac000)
	libgcc_s.so.1 => /lib/x86_64-linux-gnu/libgcc_s.so.1 (0x00007fa498491000)
	libc.so.6 => /lib/x86_64-linux-gnu/libc.so.6 (0x00007fa49829f000)
==>>>	libpcsclite.so.1 => /lib/x86_64-linux-gnu/libpcsclite.so.1 (0x00007fa498292000)
==>>>	libedit.so.2 => /lib/x86_64-linux-gnu/libedit.so.2 (0x00007fa498258000)
	libcurl.so.4 => /lib/x86_64-linux-gnu/libcurl.so.4 (0x00007fa4981c6000)
	libsrt.so.1 => /lib/x86_64-linux-gnu/libsrt.so.1 (0x00007fa498125000)
	libdl.so.2 => /lib/x86_64-linux-gnu/libdl.so.2 (0x00007fa49811f000)
	libpthread.so.0 => /lib/x86_64-linux-gnu/libpthread.so.0 (0x00007fa4980fc000)
	librt.so.1 => /lib/x86_64-linux-gnu/librt.so.1 (0x00007fa4980f2000)
...
```

Any hint / tip on what i'm doing wrong would be very much appreciated. 
Thanks
-w

lelegard, 2022-07-16:
Hi @bugre

This should not happen. This does not happen when I run the same `make` command, there is no dependency on pcsc and libedit.

Are you sure that the `/usr/local/bin/tsp` you are testing is the one that was built? Did you `make install SYSPREFIX=/usr/local` ? Did you check the file time stamps?

Alternatively, you may use some old version of TSDuck for which there could be an issue. Please try the latest version, 3.31.

bugre, 2022-07-20:
Hi @lelegard ,

Thanks for replying and sorry taking so long to getting back. 

You made very valid questions. I was sure that i did make install and was using ldd against the /usr/local/bin/tsp. But, was not sure the make install.. did install my latest build, as i could not remember having done the 'make distclean' :( . 

And, you're correct! 
There was an old build in "bin/release-x86_64....", that was build without specifying the options to make, and the 'make install..' was installing that build and not the newest one. 

Cleaned up everything, new clone from git, and build with options and it works as expected.

Thank you for helping me find my error, and sorry having wasted your time not having checked it correctly.

Have a nice day.
-wm

closed 2022-07-20 by bugre