tsduck/tsduck #1034: [Q] Generate independent PCR pid

Koipse, 2022-07-21:
Hello!

Thank you for this awesome toolkit!

I have one question: is it possible to create "independent" PCR pid from stream?

For example: i create some stream with PCR pid 256 but have some problems with him.

I want to create a new independent pid that will generate by server and replace the problem PCR old pid by new.

I saw some plugins that do smth like this, but literally didn't understand how it works?

Does anyone have any example or can me explain how it works?

Thank you!

lelegard, 2022-07-31:
Hi @Koipse 

What you want to do is insufficiently specified to provide an example.

- Do you want to regenerate new PCR values or do you want to reuse an existing PID which is known to contain better PCR values than PID 256?
- Is the TS VBR or CBR? In the VBR case, it is impossible to fix PCR since there is no way to guess the "correct" PCR value. In the CBR case, PCR can be fixed using the `pcradjust` plugin.
- Is the PID 256 a video PID with PCR or just referenced as PCR PID of the service and contains nothing but PCR's?

Koipse, 2022-08-15:
Hi @lelegard 

Thank you for reply!

The TS is CBR. Multicast. 

Yes, the PID 256 is a video PID. 

If we talk about pcradjust - this plugin is great! But had some issue with audio that out of sync after 2-3 minutes and losing after 10 minutes after starting the TSDuck. The source of this problem can't find, unfortunately. 

About this problem i wrote here: [https://github.com/tsduck/tsduck/issues/1006](url)

Koipse, 2022-08-17:
BTW the pcradjust plugin works well if you correctly count your bitrate.

Thank you!

closed 2022-08-17 by Koipse