tsduck/tsduck #1035: [Q] Get actual play-out time of TS

victorpinero, 2022-07-22:
Hello,

I would like to know if it possible to get the play-out time while running a Transport Stream.
I'm using python 3.8, Dektec board and TSDuck 3.31 - 2751.
The code used (short version) is :
`
import tsduck
TS_ASYNCREPORT = tsduck.AsyncReport(severity = tsduck.Report.Verbose, timed_log = True, sync_log = True)
TS_ASYNCREPORT.info("TSDuck version: %s" % tsduck.__version__)

TS_TSP = tsduck.TSProcessor(TS_ASYNCREPORT)
TS_TSP.input = ['file',TS_PATH, '--infinite'  ]
TS_TSP.output = ['dektec', '--device 0', [...] ]
TS_TSP.start()
TS_TSP.delete()
`

Is it maybe another way to start the TSP and control the TS information?

Thank you so much.

lelegard, 2022-07-31:
Hi @victorpinero 

What do you mean by "get the play-out time while running a Transport Stream"?

Since your output is a Dektec modulator or ASI device, the hardware regulates the playout according to the parameters of the `dektec` plugin and the playout time is simply the real time ("wall-clock time").

victorpinero, 2022-08-01:
Hi @lelegard

Sorry for the miss-explanation. 

I want to check the actual TS time that is playing into a Dektec PCI board (e.g. 5 sec of 1 min stream -> **00:05**/1:00).
I would like to know if this time can be easily taken during an infinite reproduction by using a tsp plugin, it appears on the tsp log or needs to be calculated.

What do you mean with " [...] is simply the real time ("wall-clock time"). "?
Do you mean that the playing time is the time that the board is playing the TS? If yes, I should calculate this time manually and it is not perfect. I thought it could be taken from any tsp plugin.

I do not know if I have explanied myself...

Thank you for your support!!
Glad to talk with you.

lelegard, 2022-08-01:
> Do you mean that the playing time is the time that the board is playing the TS? 

Yes, this is the definition of the play-out time. This is true in your case because the output is a regulated hardware.

> I thought it could be taken from any tsp plugin.

tsp plugins work on transport stream stuff. Computing the difference between start and stop time is not transport stream stuff.

victorpinero, 2022-08-01:
Nice explanation, kid like, I love it.

So, this time should be taken directly from the Dektec board.
I will find another way to get this time. 

Thank you so much! I did not understood the TSDuck behaviour with the board, now I do.

Thank you sir.

closed 2022-08-01 by victorpinero