tsduck/tsduck #1036: EMMG<->MUX over UDP

nikpakar, 2022-07-25:
Hello everyone, is there any option to connect tsemmg to mux over UDP port. I tried -u port option but no success,

Appreciate any feedback.

Tks
Nik

lelegard, 2022-07-25:
Hi @nikpakar 

You never _connect_ to the MUX using UDP, you always connect using TCP. All the control layer is always exchanged on the TCP session. The option `--udp` specifies that the _data_provision_ messages (eg. EMM's) are sent using UDP. This is the way the protocol is defined.

nikpakar, 2022-07-25:
Thanks Legard,

Im working on a test lab with a new production MUX from a different vendor and they only have a setting udpPort for emmChannel. Hence why i was wondering. Will double check with them.

Thanks again.