tsduck/tsduck #1037: [Q] How to change PIDs of PMT/STD/etc. via patch-xml option?

maximk-1, 2022-07-28:
Hi!

First, thank you for such good tool!

My question is

Have such tables
`$ tstables ~/file.ts --xml-output tables.xml`

```xml
<?xml version="1.0" encoding="UTF-8"?>
<tsduck>
  <PAT version="13" current="true" transport_stream_id="0x0002">
    <metadata PID="0"/>
    <service service_id="0x000D" program_map_PID="0x00F2"/>
  </PAT>
  <PMT version="1" current="true" service_id="0x000D" PCR_PID="0x00F0">
    <metadata PID="242"/>
    <registration_descriptor format_identifier="0x43554549"/>
    <component elementary_PID="0x00F0" stream_type="0x1B">
      <AVC_timing_and_HRD_descriptor hrd_management_valid="false" fixed_frame_rate="false" temporal_poc="false" picture_to_display_conversion="false"/>
      <generic_descriptor tag="0x97">
      </generic_descriptor>
    </component>
    <component elementary_PID="0x00F1" stream_type="0x03">
      <ISO_639_language_descriptor>
        <language code="rus" audio_type="0x00"/>
      </ISO_639_language_descriptor>
    </component>
    <component elementary_PID="0x00F4" stream_type="0x06">
      <teletext_descriptor>
        <teletext language_code="eng" teletext_type="0x01" page_number="100"/>
        <teletext language_code="eng" teletext_type="0x02" page_number="888"/>
      </teletext_descriptor>
    </component>
    <component elementary_PID="0x00F5" stream_type="0x86">
      <cue_identifier_descriptor cue_stream_type="insert_null_schedule"/>
    </component>
  </PMT>
  <SDT version="9" current="true" transport_stream_id="0x0002" original_network_id="0x0000" actual="true">
    <metadata PID="17"/>
    <service service_id="0x000D" EIT_schedule="false" EIT_present_following="false" CA_mode="false" running_status="running">
      <service_descriptor service_type="0x01" service_provider_name="Provider01" service_name="Channel01"/>
    </service>
  </SDT>
  <splice_information_table protocol_version="0" pts_adjustment="90,078" tier="0x0FFF">
    <metadata PID="245"/>
    <splice_insert splice_event_id="0x000093B8" splice_event_cancel="false" out_of_network="false" splice_immediate="false" unique_program_id="0x0003" avail_num="14" avails_expected="255" pts_time="7,958,685,856"/>
  </splice_information_table>
  <splice_information_table protocol_version="0" pts_adjustment="90,078" tier="0x0FFF">
    <metadata PID="245"/>
    <splice_insert splice_event_id="0x000093B8" splice_event_cancel="false" out_of_network="false" splice_immediate="false" unique_program_id="0x0003" avail_num="14" avails_expected="255" pts_time="7,958,685,856"/>
  </splice_information_table>
  <splice_information_table protocol_version="0" pts_adjustment="90,078" tier="0x0FFF">
    <metadata PID="245"/>
    <splice_insert splice_event_id="0x000093B8" splice_event_cancel="false" out_of_network="false" splice_immediate="false" unique_program_id="0x0003" avail_num="14" avails_expected="255" pts_time="7,958,685,856"/>
  </splice_information_table>
</tsduck>
```

and such patch file
```xml
<?xml version="1.0" encoding="UTF-8"?>
<tsduck>
  <PAT x-update-version="10" x-update-transport_stream_id="0x000F">
    <service x-update-service_id="0x000F" x-update-program_map_PID="0x00F3"/>
  </PAT>
  <PMT x-update-version="10" x-update-service_id="0x000F">
    <metadata x-update-PID="0x00F3" />
    <component x-update-elementary_PID="0x019a" stream_type="0x1B" />
[... 102 lines cut ...]
    <service service_id="0x000F" EIT_schedule="false" EIT_present_following="false" CA_mode="false" running_status="running">
      <service_descriptor service_type="0x01" service_provider_name="Provider02" service_name="Channel02"/>
    </service>
  </SDT>
  <splice_information_table protocol_version="0" pts_adjustment="90,078" tier="0x0FFF">
    <metadata PID="245"/>
    <splice_insert splice_event_id="0x000093B8" splice_event_cancel="false" out_of_network="false" splice_immediate="false" unique_program_id="0x0003" avail_num="14" avails_expected="255" pts_time="7,958,685,856"/>
  </splice_information_table>
  <splice_information_table protocol_version="0" pts_adjustment="90,078" tier="0x0FFF">
    <metadata PID="245"/>
    <splice_insert splice_event_id="0x000093B8" splice_event_cancel="false" out_of_network="false" splice_immediate="false" unique_program_id="0x0003" avail_num="14" avails_expected="255" pts_time="7,958,685,856"/>
  </splice_information_table>
  <splice_information_table protocol_version="0" pts_adjustment="90,078" tier="0x0FFF">
    <metadata PID="245"/>
    <splice_insert splice_event_id="0x000093B8" splice_event_cancel="false" out_of_network="false" splice_immediate="false" unique_program_id="0x0003" avail_num="14" avails_expected="255" pts_time="7,958,685,856"/>
  </splice_information_table>
</tsduck>
```

What i am doing wrong?

```shell
$ tsversion
3.31-2761
```

lelegard, 2022-07-31:
Hi @maximk-1 

> but after applying this patch to ts file - pids of PMT, SDT and splice_information_table are not changing

I think that there is a misunderstanding about what the plugins `pat`, `pmt`, `sdt` do. They only update the content of the PAT, PMT and SDT tables. If you look at the output, the three tables were correctly updated according to the patch file. But this all what those plugins were supposed to do. Specifically, there is no impact on the content of the other PID's. If you want to remap the corresponding PID's, also use the plugin `remap` (RTFM for details).

It is important to understand that each plugin does a specific job and nothing more. If you want to reconfigure the TS, you need to update the signalisation (this is what you did) and also update the or remap each individual PID. If you wonder why TSDuck let you create an invalid TS (update the signalisation without modifying the contents of the PID's), this is done on purpose. TSDuck is a low-level toolkit for tests. Creating an invalid TS by updating one part without the other is a valid test case. 

If you want a global consistent update of the TS, you need to understand all required modifications and address them one by one, possibly using several types of plugins.

Additionally, the XML element `<metadata>` is not part of a table. This is an information which is added by `tstables`. Using it in a patch file does nothing and certainly not modify the PID which carries the table.

maximk-1, 2022-08-02:
Thank you for clarification.
I hope i can make one universal patch file and make all my result files/streams with identical pids without inspecting source pids and doing manual remap.

closed 2022-08-02 by maximk-1