tsduck/tsduck #1039: [Q] Choppiness when merging multiple streams

lmoody1, 2022-08-09:
I apologize in advance for my ignorance regarding muxing and transport streams in general. I'm learning as I go.

I'm having issues trying to remap streams. In my examples, I'm attempting to remap  PID **0x1e22** to **0x1022** and PID **0x1e23** to **0x1023**.

Here is a sample of a tsp command that works:

    tsp --control-local localhost --control-port 8000 -v -b 41209008 -I null -j -O  ip localhost:8999 
    -P regulate 
    -P merge -j -t --no-pcr-restamp -b 41209008 -p 0x0-0x21 -d 0x22-0x1ffe "tsp -v -b 41209008 -I file -i C:\files\CAT_File.ts -P regulate"  
    -P merge -j -t --no-pcr-restamp -b 41209008 -p 0x1e22-0x1e23 -d 0x0-0x1e21 -d 0x1e24-0x1ffe "tsp -v -b 41209008 -I file -i C:\files\Stream_File.ts -P regulate"  
    -P remap -n 0x1e22=0x1022 0x1e23=0x1023

PIDs **0x1e22-0x1e23** are passed through and remapped **0x1e22=0x1022** and **0x1e23=0x1023**

It seems to me that the following 2 commands should generate the same net effect, but I'm not seeing any a/v output:

  PIDs **0x1e22-0x1e23** are passed through (from the second -P merge plugin) and should be remapped (from within the "tsp" command in the second -P merge plugin). Note that I've also tried this command through the remapped PIDs **0x1222-0x1223** in the -P merge plugin, with the same results:

    tsp --control-local localhost --control-port 8000 -v -b 41209008 -I null -j -O  ip localhost:8999 
    -P regulate 
    -P merge -j -t --no-pcr-restamp -b 41209008 -p 0x0-0x21 -d 0x22-0x1ffe "tsp -v -b 41209008 -I file -i C:\files\CAT_File.ts -P regulate"  
    -P merge -j -t --no-pcr-restamp -b 41209008 -p 0x1e22-0x1e23 -d 0x0-0x1e21 -d 0x1e24-0x1ffe "tsp -v -b 41209008 -I file -i C:\files\Stream_File.ts -P remap 0x1e22=0x1222 0x1e23=0x1223 -P regulate" 

  Uses -P filter and -P remap in the "tsp" command in the second -P merge plugin:

    tsp --control-local localhost --control-port 8000 -v -b 41209008 -I null -j -O  ip localhost:8999 
    -P regulate 
    -P merge -j -t --no-pcr-restamp -b 41209008 -p 0x0-0x21 -d 0x22-0x1ffe "tsp -v -b 41209008 -I file -i C:\files\CAT_File.ts -P regulate"  
    -P merge -j -t --no-pcr-restamp -b 41209008 "tsp -v -b 41209008 -I file -i C:\files\Stream_File.ts -P filter -p 0x1e22=0x1e23 -P remap 0x1e22=0x1222 0x1e23=0x1223 -P regulate" 
    
I can't see what I'm doing wrong.

I appreciate your help.

lelegard, 2022-08-10:
Hi @lmoody1 

In command number 2:
~~~
    -P merge -j -t --no-pcr-restamp -b 41209008 -p 0x1e22-0x1e23 -d 0x0-0x1e21 -d 0x1e24-0x1ffe "tsp -v -b 41209008 -I file -i C:\files\Stream_File.ts -P remap 0x1e22=0x1222 0x1e23=0x1223 -P regulate" 
~~~
In the forked process, you remap PID's 0x1e22 and 0x1e23 to 0x1222 and 0x1223. So, the parent merge plugin receives PID's 0x1222 and 0x1223. In the parent merge plugin, you drop PID's 0 to 0x1e21. So, PID's 0x1222 and 0x1223 are dropped and the merge plugin lets no PID pass.

In command number 3:
~~~
-P merge -j -t --no-pcr-restamp -b 41209008 "tsp -v -b 41209008 -I file -i C:\files\Stream_File.ts -P filter -p 0x1e22=0x1e23 -P remap 0x1e22=0x1222 0x1e23=0x1223 -P regulate" 
~~~
The syntax "-P filter -p 0x1e22=0x1e23" is incorrect and the forked process fails with an error message that you should see. If you want to filter PID's 0x1e22 to 0x1e23, the correct syntax is "-P filter -p 0x1e22-0x1e23".

lmoody1, 2022-08-11:
Hi @lelegard 

Thank you for your assistance. Clearly that was my fault. I was able to get it working.