tsduck/tsduck #1041: Problem with character interpretation in passphrase

Din-Hoking, 2022-08-17:
Hi all, 
can you please help! 
We saw when capturingSRT traffic, if there are "@ or !" characters in the passphrase, tsp sees an error
": error reading command line arguments from file "passphrase!A%D*G-"

command example:
tsp -I srt --transtype live --messageapi localhost:7086 --streamid stream_5 --passphrase '@passphrase!A%D*G-' -P until --seconds 20 -O file dump.pcap 

Similar behavior has been observed on different versions of tsduck . 
Last test was done on OS Ubuntu 2204
tsp: TSDuck - The MPEG Transport Stream Toolkit - version 3.31-2761

lelegard, 2022-08-17:
Hi @Din-Hoking 

The @ character is used to read parameters from text files. You need to double it (@@) to use a literal @. See the TSDuck user's guide section 3.1.5 for more details.

Din-Hoking, 2022-09-05:
sorry @lelegard  for the long answer, yes you are right. Thanks for the clarification

closed 2022-09-05 by Din-Hoking