tsduck/tsduck #1044: TSDuck for Raspberry Pi4 Running Ubuntu Server 20.04

picatchoof, 2022-08-20:
Hello All
it has been great experience testing TSDuck previously on my Desktop.
It was for capturing and playing around within DVB signals using TBS PCIe Cards.
Now I replying the same experience but using a RPi4 instead of Desktop. 
The signal will be captured using Geniatech HD Star USB 2.0 (SU3000-Geniatech-HDStar-v3) Card.
**Please Need the TSDuck arm64 for such OS if available.**

I get lost trying to compile from source 
Best regards.

lelegard, 2022-08-21:
Hi @picatchoof 

I do not have any arm64 distro to build TSDuck. I only have a Raspberry Pi 3 with a 32-bit Raspbian distro (armv7 build). Maybe someone else can provide an arm64 binary for Ubuntu 20. Note that TSDuck is already built on arm64 for macOS, thanks to Homebrew packager.

That being said, building TSDuck on Linux, especially on a mainstream distro such as Ubuntu, is a piece of cake. [Read this for more details](https://tsduck.io/doxy/building.html).

Once you have cloned the repo, execute the script `scripts/install-prerequisites.sh` to install all required tools and libraries to build TSDuck. Then, just `make -j8` (for 8 parallel compilations, adjust according to your memory), then `make installer` to build the `.deb` package.

Let us know the result.

picatchoof, 2022-08-21:
Hello All
I did what you said.
I takes time, I don't know how much memory it takes to do so, but it's not a big issue.
It generates the two .deb files (normal and DEV).
please find them attached the this post.

[TSDuck_Ubuntu-2004_arm64_RPi4.zip](https://github.com/tsduck/tsduck/files/9389346/TSDuck_Ubuntu-2004_arm64_RPi4.zip)

I did install one its okey
thnks a lot

lelegard, 2022-08-21:
Thanks @picatchoof. I have posted your build in the [pre-release area](https://tsduck.io/download/prerelease/). It can be useful to someone else.

The build takes time, indeed. There are about 1000 C++ source files to compile. This is why using the largest `-j` option as possible is important. On a Raspberry Pi, the performance bottleneck is usually the flash memory card, which is very slow compared to an SSD disk.

closed 2022-09-21 by lelegard

picatchoof, 2023-01-27:
Hello All
please fine attached the new TsDuck 3.34 DEB files for Raspberry 4 running Ubuntu Desktop 22
tsduck_3.34-3140.ubuntu22_arm64.
BTW I tested them my self.
[TSDuck_Ubuntu-2004_arm64_RPi4.zip](https://github.com/tsduck/tsduck/files/10518809/TSDuck_Ubuntu-2004_arm64_RPi4.zip)

lelegard, 2023-01-28:
Hi @picatchoof 

Thanks for the binaries for Raspberry Pi 4.

In the latest release, I have included some [Arm64 binaries for Ubuntu 22.10](https://tsduck.io/download/tsduck/). They were built on an Ubuntu 22.10 virtual machine running on a Mac host with an M1 chip ("Apple Silicon") and macOS 13 (Ventura). If you run the same version 22.10 on your Raspberry Pi 4, could you please try them. I am curious about the level of compatibility of those binaries.

By the way, for those who already built TSDuck on an Intel machine, the build on the M1 chip took 2mn 30s. Yes, 2, not 20...