tsduck/tsduck #1045: [Q] About SCTE-35 and pts_time

Koipse, 2022-08-21:
Greetings!

Is anyone can explain me how to correctly add the "pts_time" in the SCTE-35 xml files please?

I have next TSDuck configuration:

```
 tsp --bitrate 11580800 --buffer-size-mb 11.5 --receive-timeout 10000 \
 -P inject -b 15000 -p 0 --xml /opt/TSDuck/xml_files/test_scte_35/pat.xml=15 \
 -P inject -b 15000 -p 4096 --xml /opt/TSDuck/xml_files/test_scte_35/pmt.xml=15 \
 -P inject -b 15000 -p 17 --xml /opt/TSDuck/xml_files/test_scte_35/sdt.xml=15 \
 -P inject -b 15000 -p 16 --xml /opt/TSDuck/xml_files/test_scte_35/nit.xml=10 \
 -P inject -b 15000 -p 18 --xml /opt/TSDuck/xml_files/test_scte_35/eit.xml=10 \
 -P pmt --service 1 --add-programinfo-id 0x43554549 --add-pid 261/0x86 \
 -I null -P regulate \
 -P merge "tsp --buffer-size-mb 11.5 --receive-timeout 10000 -I ip -l ${in_iface_ip} ${input_channel} " \
 --no-psi-merge --joint-termination --no-pcr-restamp \
 -P inject --xml /opt/TSDuck/xml_files/test_scte_35/tdt.xml=10000 /opt/TSDuck/xml_files/test_scte_35/tot.xml=10000 -b 15000 -p 20 \
 -P spliceinject --service 1 --files /opt/TSDuck/xml_files/test_scte_35/scte-35.xml -p 261 --max-file-size 4096  --min-bitrate 250  \
 -P timeref --system-synchronous \
 -P pcradjust \
 -P continuity --fix \
 -O ip -l ${out_iface_ip} --ttl 128 ${output_channel}

```

Here is where i inject the SCTE-35 marks:

` -P spliceinject --service 1 --files /opt/TSDuck/xml_files/test_scte_35/scte-35.xml -p 261 --max-file-size 4096  --min-bitrate 250 `

The SCTE-35 xml file:

```
<splice_information_table>
 <splice_insert splice_event_id="105" unique_program_id="1"
                out_of_network="true" pts_time="1660838220"
                splice_immediate="false"/>

 <splice_segmentation_descriptor segmentation_event_id="105" segmentation_type_id="0x24" segment_num="1" segments_expected="1">
 <segmentation_upid type="0"/>
 </splice_segmentation_descriptor>

 </splice_information_table>

 <splice_information_table>
 <splice_insert splice_event_id="105" unique_program_id="1"
                out_of_network="false" pts_time="1660838220"
                splice_immediate="false"/>

 </splice_information_table>

 <splice_information_table>
 <splice_insert splice_event_id="106" unique_program_id="2"
                out_of_network="true" pts_time="1660838280"
                splice_immediate="false"/>

 </splice_information_table>

 <splice_information_table>
 <splice_insert splice_event_id="106" unique_program_id="2"
                out_of_network="false" pts_time="1660838340"
                splice_immediate="false"/>

 </splice_information_table>
```

And if understand correctly, i can use the epoch timestamp in pts_time to create the SCTE-35, right? Or this point need another variable? 

For example, i need to put the SCTE-35 to the multicast stream from 21.08.2022 21:05 to 21.08.2022 21:10

So the pts_time will be:

1. CUE_OUT = 1661115900 (21:05)
2. CUE_IN = 1661116200 (21:10)

I check this marks by Bridgetech.
But if i will check this stream in this time will not see any event id or smth like if we put the splice_immediate="true". 
Yea i know that the splice_immediate="true" add the SCTE-35 immediately after we change the xml file.

I'm afraid that don't correctly understand how the "pts_time" works and did wrong.

Thank you for advice!

lelegard, 2022-08-22:
Hi @Koipse 

The PTS is defined in ISO/IEC 13818-1 and H.222.0 as a time stamp with a granularity of 1/90000 second (ie. at 90 kHz) on 33 bits. This means that the PTS value wraps at zero approximately every 26 hours.

There is no PTS epoch. There is no PTS synchronization reference. The PTS is based on the PCR of the service (PTS=PCR/300). Each service uses its own PCR (program clock reference).

So, the value of the PTS for a given anticipated time is bound to the current service. You need to know the current PTS of the service you target and then compute a PTS in the future by adding the number of 1/90000 second.

Koipse, 2022-08-22:
Awesome! 

Thank you for this information!

It helped me to understand how to correctly work with SCTE-35.

closed 2022-08-22 by Koipse